mavlink/qgroundcontrol #2636: Joystick should not be in APM

DonLakeFlyer, 2016-01-16:


DonLakeFlyer, 2016-01-16:
@LorenzMeier Any ideas where given #2642 which has MANUAL_CONTROL support on custom APM stack. The issue is that Joystick shows a lot of UI that may not work correctly on all firmwares. But whether joystick support is there or not can't really be trigggered on firmware type. Needs to be triggered some other way.

DonLakeFlyer, 2016-01-16:
New values for MAV_PROTOCOL_CAPABILITY?

DonLakeFlyer, 2016-01-16:
@rjehangir Adding you, so you can follow along on this one.

rjehangir, 2016-01-18:
Don, thanks. I want to chime in on this issue because it's important for our work. We're developing ArduSub (https://github.com/bluerobotics/ardupilot) and relying primarily on real and virtual joystick in QGC for user control input. 

We are happy to contribute to and test the joystick thoroughly to make sure that it functions correctly.

closed 2016-01-21 by DonLakeFlyer