mavlink/qgroundcontrol #2638: Tame the CPU hog

LorenzMeier, 2016-01-16:
@DonLakeFlyer I know this is not fun and not so much a consideration during desktop development, but its the No. 1 complaint in the field: QGC eating your battery alive. This is what it currently looks like for me.

It ramps up my fan quite a bit and I'm relatively sure that my field battery life would not be great.

[image]

LorenzMeier, 2016-01-16:
Note that I included WindowServer because its load goes from single-digit percent to 35 once QGC starts.

DonLakeFlyer, 2016-01-16:
Unless there is something trival here I don't think this will fit into 3.0 release.

LorenzMeier, 2016-01-16:
It might make sense to run a function profiler to see if the load is somewhere it can be easily fixed, or if its across the board. I had QGC around 20-25% back in the day when I regularly profiled it.

LorenzMeier, 2016-01-16:
Note that native QML main Window might also help because it uses different rendering - didn't you work out recently how to do this?

DonLakeFlyer, 2016-01-16:
Hmm, maybe. I'll still have a branch for that. I'll try it to see if it makes a difference. Another interesting thing is that in the commercial version of Qt there is a Qml compiler.

DonLakeFlyer, 2016-01-17:
What sort of connections were you running when you did this test? We can't get anything higher than 20% CPU to show up.

NaterGator, 2016-01-19:
@DonLakeFlyer The Qt Quick compiler should only help with initial load times; once the JIT has compiled at runtime things should theoretically be the same. The Qt Quick [performance guide](http://doc.qt.io/qt-5/qtquick-performance.html#bindings) is worth looking over, but you've probably already seen it. 

@LorenzMeier any chance you were streaming video via QGC/Gstreamer?

DonLakeFlyer, 2016-01-21:
I'm planning to move this to 3.1 release. Speak up now if you disagree.

DonLakeFlyer, 2017-04-06:
Current builds seem to be ok

closed 2017-04-06 by DonLakeFlyer