mavlink/qgroundcontrol #2649: Log Download for APM

lvale, 2016-01-16:
Some more options for APM vehicle Log download:

Currently it seems that we have to do it one at a time and there is no feedback. 

[image]

I would suggest something more like what APM Planner has:

[image]

[image]

DonLakeFlyer, 2016-01-16:
Makes sense

dogmaphobic, 2016-01-16:
It's all there. Expand the window to see the other fields and select any number of files for download (Command-A for everything).

P.S. It must *rain* a lot in **W**estern **E**uropean **T**ime :P

[image]

lvale, 2016-01-16:
Thanks @dogmaphobic. Usable but no intuitive. I didn't think of resizing the dialog box, and the multiple selection is obvious AFTER we know about it :)

https://www.youtube.com/watch?v=gL0CzPQ0pAQ

Now those disconnects and reconnects kill me :)

dogmaphobic, 2016-01-16:
Those voice messages are basically useless and extremely annoying! As to the link lost/regained, how are you connecting?

dogmaphobic, 2016-01-16:
Closing as the issue itself is not valid. The crazy link connect/disconnect should be a new issue.

lvale, 2016-01-16:
Please consider resizing the widget on first time it opens.

dogmaphobic, 2016-01-16:
> Please consider resizing the widget on first time it opens.

It should have a sensible default. Obviously that's not working. Let me reopen and deal with that then.

lvale, 2016-01-16:
If I may suggest to have the column Status as First column and the ID either don't show it or have it as last column.

dogmaphobic, 2016-01-16:
If you don't have GPS, the dates are all the same. The index is all you have to go by.

LorenzMeier, 2016-01-16:
I don't think it makes sense to spend a lot of time on the widget, as we still don't have it on tablets then - shouldn't a fix maybe be done straight in QML?

dogmaphobic, 2016-01-16:
It's all QML. Just the *container* is a widget.

DonLakeFlyer, 2016-01-16:
> It's all QML. Just the container is a widget.

Ooh, that might be a problem if I try to do the Native QMl thing on desktop. In that case I think the only Qml can be main window. Hmm.

dogmaphobic, 2016-01-17:
#2654 takes care of the connection loss.

lvale, 2016-01-18:
Thanks @dogmaphobic

closed 2016-01-18 by lvale