mavlink/qgroundcontrol #2650: Android QGC

lvale, 2016-01-16:
Latest beta on Android tablet.

Screen goes off. I would suggest an option to force screen on settings.

Also no widgets on Android (I was trying to download logs from vehicle)

DonLakeFlyer, 2016-01-16:
Widgets on Android is going to take some time. Do you think Log Download is critical on tablet? You won't be able to do anything other than pull it off the tablet to a desktop.

lvale, 2016-01-17:
No hurry for the widgets. The log download is important while you're configuring a vehicle and we must examine the log, like to set the value to TH_MID. Although I still use the Mac on the field :)

The screen turning off is more annoying.

DonLakeFlyer, 2016-01-18:
V3.0 for screen going Off. 3.1 for the rest.

DonLakeFlyer, 2016-01-21:
@dogmaphobic Can you take a look at this. Also not sure if WAKE_LOCK is correct: http://stackoverflow.com/questions/4195682/android-disable-screen-timeout-while-app-is-running

DonLakeFlyer, 2016-06-29:
@dogmaphobic Maybe one more stab at this for Android and iOS, then give up till 3.1? Android more important that iOS since it will move to Stable with 3.0.

dogmaphobic, 2016-06-29:
iOS is working (has been for quite a while). I could not find a way to make it work on Android.

DonLakeFlyer, 2016-06-29:
Moving to 3.1 then

closed 2017-01-02 by lvale