mavlink/qgroundcontrol #2656: Change reference for Altitude in the HUD view

pkocmoud, 2016-01-17:
Without a GPS the Altitude reads 0. 

Can we change the reference in the QGC for altitude to VFR_HUD : alt ?

[image]

LorenzMeier, 2016-01-17:
@DonLakeFlyer @dogmaphobic When using non-AMSL (from GLOBAL_POSITION_INT) the altitude indicator should be e.g. red and have these two indications:

With global position: Altitude (AMSL m)
Without: Altitude (baro m)

We need to make sure people understand what they are seeing and using.

DonLakeFlyer, 2016-01-17:
Can't it just be (m) and (baro m)?

LorenzMeier, 2016-01-17:
It can, but then its not clear if this the altitude over ground or absolute (in particular in your area where these are pretty close).

LorenzMeier, 2016-01-17:
I guess I'm fine whatever works for users, as long as baro altitude is flagged somehow.

dogmaphobic, 2016-01-17:
Make it **GPS Altitude** and **Barometer Altitude** accordingly?

DonLakeFlyer, 2016-01-17:
Just trying to keep unneeded clutter off the UI. Let's just flag baro. If we hear there is confusion we can add.

DonLakeFlyer, 2016-01-17:
We're not switching to VFR_HUD are we. Not sure that works on APM stack.

pkocmoud, 2016-01-18:
What is the purpose of all of the duplicate MavLink messages, seems like a great time to audit them?

DonLakeFlyer, 2016-05-13:
Two issues:
1) This may be a little more complicated with the new Vehicle object model support.
2) Which mavlink message should the information come from in which cases? Need to do a review of the Vehicle object model and what messages the values are coming from.

sujiv0, 2017-10-06:
How to include Throttle Percentage Value on to the HUD/Widget??
Please add this feature too @DonLakeFlyer  @LorenzMeier

closed 2026-10-04 by github-actions[bot]