mavlink/qgroundcontrol #2663: Ask user to put the SD card back in if connecting to an unresponsive Pixhawk or Pixracer

LorenzMeier, 2016-01-18:
APM and PX4 now both boot into "debug mode" with the SD card removed. QGC will be able to open the link, but not communicate with the board in that state. If that happens, it should ask the user to put a microSD card in.

DonLakeFlyer, 2016-01-18:
Detected how?

LorenzMeier, 2016-01-19:
Essentially you have a Pixhawk or Pixracer detected, but its not sending any data back (or it might actually return data on heartbeats, but no MAVLink).

LorenzMeier, 2016-01-20:
People struggling:
http://discuss.px4.io/t/pixracer-feedback-thread/27/28

DonLakeFlyer, 2016-01-20:
> Essentially you have a Pixhawk or Pixracer detected, but its not sending any data back (or it might actually return data on heartbeats, but no MAVLink).

The current pattern is here:
- QGC detects incoming heartbeat, starts a new Vehicle. So heartbeats are coming through.
- QGC sends outgoing parameter request and waits for response
- If QGC times out on parameter request response it pops a message to the user and retries

That's all I have to go on. I can change the message that pops up when a parameter requests fails to also say "possibly you don't have the SD card in", but parameter requests fail in other cases as well. So it's only a "maybe" that's the problem. Unless there is some other more direct way to determine this case.

LorenzMeier, 2016-01-20:
There is a direct way: You never get a heartbeat. Not a single MAVLink message. No vehicle. Just try with a Pixhawk without microSD inserted.

DonLakeFlyer, 2016-01-20:
Ok, so it needs to know that it has a direct usb connection to a Pixhawk board and it never gets a heartbeat. Doable.

DonLakeFlyer, 2016-01-21:
@LorenzMeier How many seconds to wait before this times out? Board boot is 5 seconds I think, so maybe something like 7 seconds?

LorenzMeier, 2016-01-21:
Sounds fair

closed 2016-01-21 by DonLakeFlyer