mavlink/qgroundcontrol #2666: QGC OS X latest

lvale, 2016-01-18:
Hi,

Again on the maps position :).

Considering that the OS has location services (at least OS X, IOS and Android have), couldn't QGC query the base OS for location and display the map if the OS reports a valid location, perhaps using a option for "Use current location for Map Center" to enable that, similar to the Center Map on Vehicle option.

DonLakeFlyer, 2016-01-18:
If the OS supports map location then that is what happens now. Location is only support nu Qt in Android and iOS at this point.

lvale, 2016-01-18:
ok, although that's why I haven't seen any requests by QGC to OS X location services:

[image]

DonLakeFlyer, 2016-01-18:
Closing this for now, since QGC already supports this. Boot up android tablet with no vehicle and map will move ot tablet location.

closed 2016-01-18 by DonLakeFlyer