mavlink/qgroundcontrol #2669: QGC - Emergency Stop on APM

lvale, 2016-01-18:
APM has implemented the concept of a "Kill Switch" on recent fw versions.

Tower has implemented that, but requires IMHO too many clicks and drags to be effective.

I would suggest a simple slider at the top to provide such functionality. If we want a 2 step verification after the user slides the control, a pop-up would appear with confirmation YES/NO.

[image]

DonLakeFlyer, 2016-02-28:
Does ArduPilot support this through some sort of MAV_CMD?

lvale, 2016-02-28:
There's no Mavlink message to do Kill. I believe the workaround used by Tower and described here might be a viable option
http://copter.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#copter-2

DonLakeFlyer, 2016-02-29:
Doesn't seem like an unreasonable extension to arm/disarm. @LorenzMeier Can we make this a standard?

LorenzMeier, 2016-02-29:
Sure, we've used that in the past and still have all the plumbing in for it. @lvale Can you send a PR against the MAVLink repo for the magic and docs?

LorenzMeier, 2016-02-29:
@DonLakeFlyer Just checked, fully implemented in PX4.

LorenzMeier, 2016-02-29:
I would say before implementing this feature flicking to pause / hold should be done in a really intuitive way. This should only come after that.

LorenzMeier, 2016-02-29:
And it should not be called emergency stop. It should be called emergency crash or something like that making clear this is something that is fatal.

lvale, 2016-02-29:
Hi @LorenzMeier . So, no need for a PR? My suggestion above of how it should work might be reasonable, because we want the feature easy, but secure to use, so the slider and a big red pop-up could cover both requisites.

Oh, and the name. Don't say crash, Just call it "EMERGENCY STOP" or "EMERGENCY", and on the pop-up that should come after the slider have one bigger green button with NO and a smaller one with YES, and some text saying something like "The action will crash your vehicle if in movement by stopping all the motors. ARE YOU SURE?"
Note that the use of Caps is on purpose.

DonLakeFlyer, 2016-02-29:
The PR is for mavlink docs update to make this a standard.

DonLakeFlyer, 2016-03-01:
PX4: EXTENDED_SYS_STATE will tell me landed/in air
ArduPIlot: MAV_STATE will tell me landed/in air

DonLakeFlyer, 2016-12-29:
Feature added a while ago

closed 2016-12-29 by DonLakeFlyer