mavlink/qgroundcontrol #2671: Crashing on startup on Windows 7 and 10 (2.8)

julianoes, 2016-01-18:
A user has [reported](http://discuss.px4.io/t/qgrountcontroll-neuste-stabile-versiob/95/5) that his QGC crashes on startup on Win7 and Win10 and only works in the XP compatibility mode. 

[image]

NaterGator, 2016-01-19:
Can you get more information about the machine it is crashing on? CPU, GPU, etc?

DonLakeFlyer, 2016-01-20:
Can you get a crash dump: http://www.qgroundcontrol.org/bugreporting?

DonLakeFlyer, 2016-01-22:
Also check the graphics driver. Intel graphics drivers are known to crash in OpenGl.

julianoes, 2016-04-28:
No more information was provided by the user, closing.

closed 2016-04-28 by julianoes