mavlink/qgroundcontrol #2678: Flash dialog is super confusing

LorenzMeier, 2016-01-20:
@DonLakeFlyer @dogmaphobic What QGC is trying to say is here is:

"UNPLUG THE USB PLUG TO START FIRMWARE UPGRADING"

I think that is what it should say. In the center. BIG. Right now users think "it doesn't work".

[image]

Then on to the 2nd issue: If you unplug, the UI resets and moves back to the overview. If you now re-plug without toggling back to Firmware upgrade you're back to square 1. Super annoying.

I have quite a few user reports on this. We need to fix it this time. I've raised it for a very long time now.

[] Fix flash instructions. Be SHORT
[] Do not reset UI but stay in Firmware upgrade view

DonLakeFlyer, 2016-01-20:
Tell me exactly word for word what you want it to say and I'll change it!

LorenzMeier, 2016-01-20:
"UNPLUG THE AUTOPILOT USB CABLE TO START FIRMWARE UPGRADING"

In center. no other text. In a big font.

DonLakeFlyer, 2016-01-20:
What about Radio?

LorenzMeier, 2016-01-20:
Right. Change it to board or something.

DonLakeFlyer, 2016-01-20:
> I've raised it for a very long time now.

This is the first I've heard of the UI reset issue. Althought now that I know about it I can certainly see it as completetly screwed up. If things just fly by in Gitter I lose them. A detailed GitHub Issue and I won't miss things.

closed 2016-01-21 by DonLakeFlyer