mavlink/qgroundcontrol #2695: Disconnect button is not close to connection lost label, "doesn't work" if not hitting it

LorenzMeier, 2016-01-22:
If you don't click disconnect after a firmware upgrade (where you pull out the cable because the UI tells you to and the link becomes stale) the board will never show up.

Because the stale link is preventing the rebooted board to come live again. The disconnect button should be close to the CONNECTION LOST label.

LorenzMeier, 2016-01-26:
Confuses more users. The issue is to do auto-connect but not auto-disconnect. It just looks broken to users because they don't understand that they need to do something.

DonLakeFlyer, 2016-01-26:
> but not auto-disconnect. 

Huh? Don't understand what you means by that. The detection is that there are no longer heartbeats coming from the vehicle. There is not auto-disconnect.

LorenzMeier, 2016-01-26:
There is no workflow for connect. So users don't have the mental model of a connection state and hence do not consider the disconnect button worth clicking. However, if they don't, a rebooted PIXHAWK will never ever connect again. So from th user perspective everything breaks after Firmware upgrade. Nothing works. The autopilot is dead.

DonLakeFlyer, 2016-01-26:
Understood. That just goes back to firmware upgrade needs a complete re-work.

DonLakeFlyer, 2016-01-26:
Auto-Connect/Disconnect may need some more explanation in general as well even if you take firmware upgrade out of the picture.

quentin1002, 2016-01-27:
Hi, 
You answered me on the #933. After the firmware's update, I clicked Disconnect because a Connection Lost Message was shown. 
Nevertheless, nothing happens now: 
- big LED is still blinking red
- Message at the bottom saying: Your Vehicle is not responding. If this continue check if you have a SD card... I do have the SD card inside, the one that is sold by 3DR. 

What should I do?

DonLakeFlyer, 2016-02-10:
@quentin1002 My guess is that you have something going wrong on the firmware side. I'd direct your questions there. If you continue to get Connect Lost when connecting to your vehicle it means QGC is not getting a heartbeat from the vehicle.

DonLakeFlyer, 2016-02-10:
See pull for usabillity change

closed 2016-02-10 by DonLakeFlyer

quentin1002, 2016-02-17:
At the end of the day, I tried to install QGC on my Windows computer and it worked. Back on my Ubuntu, I realized that I had to unplug all my USB devices. And now QGC also works on my Ubuntu computer. The lack of USB energy might be the reason of the connection lost error...

DonLakeFlyer, 2016-02-17:
Great, thanks reporting back. Yes you need to plug in directly to the computer to get full power.