mavlink/qgroundcontrol #2700: Plan View: Need to figure out fake home position usage while editing

DonLakeFlyer, 2016-01-22:
Current state:
- ArduPilot firmware expects a fake home position in position 0 of waypoints. This is true when load/save to vehicle or file.
- PX4 firmware does not have a fake home position anywhere.
- AP2/MP uis support the concept of fake home position while editing
- QGC currently handles the firmware differences for fake home position with respect to the back end
- QGC does not support the concept of fake home position while editing
- QGC only supports showing home position which comes from an active vehicle

Issues:
- When offline editing, lack of home position doesn't show you the full visualization of the mission for start/end of mission if you start from the ground and/or do an RTL since home is missing.
- When offline editing, lack of home position causes the atitude map to not be very useful in cases where you are mixing relative/non-relative alt.
- When online editing the active vehicle home position with respect to a mission is only correct when you are actually at the location you will fly the mission. 
- When online editing QGC will connect waypoint lines to the active vehicle home position, which in many cases will be an incorrect visualization of the mission.
- ArduPilot users will be confused when moving to QGC with the home position editing difference

All of which leads me to believe:
- QGC needs to support a full and as correct as possible visualization of a mission in both offline and online modes
- QGC needs to support the concept of a fake home position in order to do that
- The user must decide whether to use a fake home position or the vehicle home position while editing
- This needs to be presented in a way that doesn't confuse the hell out of users which will be tricky

DonLakeFlyer, 2016-01-22:
Keep in mind that I am only talking about editing experience here. I make no comments about a fake home position actually being on the Vehicle. I expect that to stay the same and is already dealt with.

DonLakeFlyer, 2016-01-22:
The fix for this will likely come with a change to json format for mission file.

DonLakeFlyer, 2016-01-22:
As I get deeper into this, the concept of storing a "Planned Home Position" on the Vehicle now starts to make sense to me. Without it, you can't read a Mission from a Vehicle and have it show a full Mission visualization with a start and finish from Home.

DonLakeFlyer, 2016-01-23:
> have it show a full Mission visualization with a start and finish from Home.

That is only the case if the Mission starts/ends on the ground which you can't quite determine without asking the user. Hmm.

DonLakeFlyer, 2016-01-23:
Work in progress can be found here: https://github.com/DonLakeFlyer/qgroundcontrol/tree/MissionHome

How it works:
* Plan View
  * Always shows a "Planned Home Position", whether offline or online and whether the vehicle has sent a true home position or not. This home position is identified clearly as "Planned" home position such that it is not confused with the actual vehicle home position.
  * The Plan View never shows home position from the Vehicle, only Planned Home Position
  * You can move this planned home position to where you want, drag, set lot/lon/alt manually, including move to the map center using the button in the item editor
  * If you load an ArduPilot mission file (V110), or load a mission from an ArduPilot Stack Vehicle, the planned home position is taken from mission item 0 which is the norm for ArudPilot users
  * If you load a QGC mission file (V120), or load a mission from a PX4 Flight Stack Vehicle, the planned home position is not available in the file/vehicle and is instead automatically generated in the center of all waypoints
  * When you save to a file it is written out as a new version (V130) which includes the planned home position in position 0. WIP in progress is to move to json format for V130 files. Json format will allow for additional features such as rally points and so forth to be saved in mission files.
  * Reading V130 mission files loads the planned home position from mission item 0 of the file
  * When a mission is loaded from a file/vehicle  the map will automatically move/zoom to fit the waypoints
* Fly View
  * Only shows actual home position from vehicle when available
  * Updates the home position as the vehicle sends new home positions
  * If there is a mission present on an active vehicle and the vehicle has not sent a home position, then the home position is not shown and any waypoint lines from the home position to the next waypoint are also not shown. When the vehicle finally sends a home position these will show up automatically

These changes allow for a full visualization of a mission from start to finish in all cases, offline/online, online with vehicle not at location the mission is planned for, and PX4/ArduPilot firmwares.

[image]

DonLakeFlyer, 2016-01-23:
WIP json format:
```
{
    "items": [
        {
            "autoContinue": true,
            "command": 22,
            "coordinate": [
                47.633263249999999,
                -122.08954304,
                25
            ],
            "frame": 3,
            "id": 1,
            "param1": 0.26179939000000002,
            "param2": 3,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        },
        {
            "autoContinue": true,
            "command": 16,
            "coordinate": [
                47.633588590000002,
                -122.08865790999999,
                25
            ],
            "frame": 3,
            "id": 2,
            "param1": 0,
            "param2": 3,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        },
        {
            "autoContinue": true,
            "command": 16,
            "coordinate": [
                47.63285114,
                -122.08894223,
                25
            ],
            "frame": 3,
            "id": 3,
            "param1": 0,
            "param2": 3,
            "param3": 0,
            "param4": 0,
            "type": "missionItem"
        }
    ],
    "plannedHomePosition": {
        "autoContinue": true,
        "command": 16,
        "coordinate": [
            47.633033699999999,
            -122.0879364,
            3.5
        ],
        "frame": 0,
        "id": 0,
        "param1": 0,
        "param2": 3,
        "param3": 0,
        "param4": 0,
        "type": "missionItem"
    },
    "version": "1.0"
}
```

DonLakeFlyer, 2016-02-09:
These changes also need to handle JUMP to id's correctly.

DonLakeFlyer, 2016-02-09:
Correctly handles loading the following file formats:
- WPL 100 - MP/AP2 format which includes home position in index 0
- WPL 120 - QGC format which does not have a home position in the file
- New .mission json format file - Described above

closed 2016-02-10 by DonLakeFlyer

billbonney, 2016-02-11:
see https://github.com/diydrones/apm_planner/blame/master/src/uas/UASWaypointManager.cc#L510

For AP2 waypoint version being 110, same as MP