mavlink/qgroundcontrol #2704: Unreliable SD Card detection

lvale, 2016-01-24:
Hi.

This new feature is giving me the absent SD when I plugin the PixHawk on APM 3.4dev, although it can reach the card to retrieve logs....

I tested in several ways.

PixHawk off: Plugin USB cable - warning appears almost immediately (less than 5 seconds)

PixHawk powered via Power module and already ready (boot process complete verified by the leds): Same thing

PixHawk off: QGC not launched, Plugin USB cable wait for boot to complete, Launch QGC, eventually detects the PixHawk and the SD warning appears almost immediately.

[image]

DonLakeFlyer, 2016-01-24:
It takes way longer to get the first heartbeart from the ArduPiliot stack than the PX4 Stack. The timeout is too short. You can disregard the error though and everything will be fine.

DonLakeFlyer, 2016-01-24:
Thanks for the report @lvale

closed 2016-01-24 by DonLakeFlyer