mavlink/qgroundcontrol #2709: Win7: Reinstall does not reset settings

LorenzMeier, 2016-01-25:
@DonLakeFlyer It seems like a re-installation on Win7 does not reset all settings:
http://fpv-treff.de/viewtopic.php?p=103983#p103983

It also looks like we should be defaulting Windows to angle maybe? @dagar Any insights / suggestions?

Its so common that we will want to sort this before 3.0.

dagar, 2016-01-25:
What's missing here? Not clearing the installation directory in program files, QGroundControl user settings, or something else?

DonLakeFlyer, 2016-01-25:
Should prompt to uninstall if you try to install over another install. Put it in about a month ago

DonLakeFlyer, 2016-01-26:
> It also looks like we should be defaulting Windows to angle maybe?

No we shouldn't. We need to blacklist built-in Intel Graphics cards such that they default to Angle. The rest work fine. There is a blacklist mechanism to do this in Qt supposedly. Was waiting to see if the Angle thing actually worked as a work-around which it seems it does. Also not sure at this point if it requires additional DLLs to be included in the install, which I think it may.

DonLakeFlyer, 2016-01-26:
> It seems like a re-installation on Win7 does not reset all settings:

What exactly is this forum post saying is going wrong?

NaterGator, 2016-01-28:
@DonLakeFlyer the qt packaging app already includes the necessary DLLs, nothing to do there. 

@LorenzMeier I don't think so. If we find out there are other OpenGL implementations that are problematic we can define our own blacklist and set QT_OPENGL_BUGLIST to override the built in one... the syntax is pretty easy: http://code.qt.io/cgit/qt/qtbase.git/plain/src/plugins/platforms/windows/openglblacklists/default.json?h=v5.5.1

That notwithstanding I wouldn't rush to blacklist drivers (bad as the state of windows OpenGL is)... I frequently find rendering quirks are caused by inappropriate transformations being applied (floating point geometry, etc). The overhead for using angle is likely not too severe, but if we have to fall back to swrast opengl we're going to have a battery hit. 

For the glitch this user is seeing: I'll submit a PR in just a bit to remove the the locally stored settings on uninstall which will fix the airframe vector graphics not displaying properly. The root cause here is he is loading a PX4AirframeFactMetaData.xml cache from his QSettings path which doesn't match the update qresource layout. He may also be having issue with SVG rendering which could stem from an incompatible Qt build in his PATH that is messing with the qt plugin resolution process.

LorenzMeier, 2016-01-29:
Fixed on master and maintenance release v2.9.2 out, thanks!

closed 2016-01-29 by LorenzMeier