mavlink/qgroundcontrol #2711: Connection and firmware upgrade issues in 2.9.1

mysid, 2016-01-26:
I've noticed some connection issues where qgc is unable to connect to my pixhawk over usb, presenting with following error messages:

[image]

[image]

[image]

However, if I plug in the 3DR radio first (with the pixhawk still unpowered and unconnected), select the firmware tab (at which point I'm faced with a prompt to unplug my USB connection to the pixhawk), unplug the radio, and then plug in the pixhawk's USB connection, the link works and the firmware upgrade process seems to work fine.

I've noticed the same issues on windows and linux (though I didn't try the plugging in the radio first and unplugging it thing as mentioned above on linux).

I do not observe any of the same connection issues on QGC v2.8 though, so I think it might be an issue with 2.9 (both .0 and .1)

NaterGator, 2016-01-29:
I'm fighting a lot of USB communication issues on windows too. 

In my experience I think at least some aspect of it is down to buggy USB CDC ACM. I'm not sure if the fault lies in the NuttX ACM driver or the usbser driver in Windows... but somehow opening and closing the COM port triggers a state where the port is still "valid"  (no WinAPI functions return errors) but there is no data flowing through. This is going to require further exploration and will likely be a pain to diagnose.

Another consideration is the connection delays imposed by waiting for Windows to initialize the driver for the bootloader, then unload it and reload for the autopilot firmware. I'm thinking we should consider _not_ having the bootloader utilize the same usb product id as the autopilot firmware. If we want to encode information in the product ID (hardware revision, etc) we should consider setting a flag (0x8000?) to indicate if the connection is coming from the bootloader or the autopilot stack.

NaterGator, 2016-01-29:
On second thought this may not be so bad.

@LorenzMeier can you take a look at [this](https://groups.yahoo.com/neo/groups/nuttx/conversations/topics/3514) thread? [This post](https://groups.yahoo.com/neo/groups/nuttx/conversations/messages/4683) looks like a smoking gun for the connection issues we're seeing in Windows. They've fixed it upstream, so if you agree I think it would be appropriate to pull in the usbdev fixes. (I suspect the best course of action is just to backport the usbdev changes since PX4/NuttX is effectively forked)

DonLakeFlyer, 2016-02-08:
Any update on this?

LorenzMeier, 2016-02-08:
These should have been back ported already. You can however run recent NuttX now as well: https://github.com/PX4/Firmware/pull/3131

If the issue goes away with that Firmware version then it is indeed NuttX.

NaterGator, 2016-02-08:
I have been having intermittent issues and have been running with a JTAG debugger attached. Most often when I notice the CDC ACM driver stops responding interrupts on the STM32 are also not function and the PC is at https://github.com/PX4/NuttX/blob/3009fb230ec8232c4c4fd7798fd70ea9632692ec/nuttx/arch/arm/src/stm32/stm32f40xxx_dma.c#L625 when I break. 

I audited upstream nuttx's otgfs code for the STM32 and there hasn't been anything that is clearly attributed to this. That said I still think it is some timing quirk with the nuttx CDC ACM driver that is incompatible with usbser in windows.