mavlink/qgroundcontrol #2715: Improve mission feedback

AndreasAntener, 2016-01-28:
Mission states are reported via various messages. One problem on the PX4 side is that the message severity sometimes make no sense. The other issue is that in QGC you need to check the messages.

A few problematic cases should be handled better, e.g. mission not accepted, takeoff too far away, etc. I'm not actually sure what is done how at the moment in terms of notifications.

@sanderux

sanderux, 2016-01-28:
Thnx, i'll see if i can improve this.

AndreasAntener, 2016-01-29:
@DonLakeFlyer @LorenzMeier do the message severity level currently play a role in communication with QGC?

LorenzMeier, 2016-01-29:
Yes, the severity controls what is spoken and what not. Anything at critical or higher will be spoken.

AndreasAntener, 2016-01-29:
Oh, wasn't there that thing with the # (hash) in front of the message for that? For the FW stuff we used `mavlink_log_info(_mavlink_fd, "#Takeoff on runway");` and I think QGC speaks that as well

LorenzMeier, 2016-01-29:
That's deprecated.

DonLakeFlyer, 2016-01-29:
The message severity determine whether the toolbar icon turns to a yield sign. Also messages above a certain severity cause the message to show in a popup at the bottom of the screen. Unless the severity trigger threshold is incorrect in QGC (or you think the toolbar/popup mechanism is insufficient) the changes needed to make this better will come from the firmware side, not the QGC side.

sanderux, 2016-01-30:
My proposal for mission feedback and analysis would be the following;

1. Revisit the mavlink log feedback. Currently a mission always raises a critical alert that states the takeoff altitude. This results in less attention to true critical alerts that could contain safety or feasibility warnings.
2. There is currently no GCS feedback on the result of uploading a mission to the FC. I have recently experienced confusion that resulted from an apparent failure to send the mission to the FC. I would propose that QGC indicate both success and failure of sending the mission. The latter i think should be a very clear warning.
3. While tracking a mission through radio telemetry it would provide a lot of clarity to see which mission item is currently being executed. This could simply be provided as mavlink log messages (info), but ideally it would be displayed in the "FLY" window of QGC. I think it now changes the color on the current mission item but in the field this is very hard to see. If the mission list would be displayed as it is in the "PLAN" window, and it would follow the currently executed item, this would provide more clarity. Not only on the sequence number, but also what that command actually is.
4. Post flight analysis of a mission can now only be performed if telemetry was connected and saved from QGC. From the onboard logs there is no way to tell (AFAIK) what mission item was being executed, or even what the mission was. I would propose that the mission is added to the preflight log, and the current mission item index added to the flight log. If mid-flight mission changes should also be considered perhaps a separate mission log is an option. 

I realize points 1 and 4 are on the PX4 side, and 2/3 on QGC side but this would be my take on the user perspective of mission feedback and analysis.

DonLakeFlyer, 2016-01-30:
#2 QGC will display an error if send to vehicle flails, with details of what went wrong. Nothing currently on a successful send.

LorenzMeier, 2016-01-30:
There should be no notification on successful sends. That's meaningless and does not represent modern user interface design. If something goes wrong it will tell you.

AndreasAntener, 2016-01-30:
I generally agree Lorenz, but in this case it should at least show the upload progress. Ok to be fair, I hear the buzzer when it's accepted and the check passed. But if I upload a big mission and don't have the buzzer (or the plane is already in the air) I have no notion of what's going on.

DonLakeFlyer, 2016-01-31:
The mission controller has an in progress signal internally. My original plan was to animate/rotate the sync icon in the button when send/recevie was in progress. I can change that when I get back.

sanderux, 2016-02-01:
@DonLakeFlyer 
From the dev call we also got the request to support user configurable severity for spoken messages.

DonLakeFlyer, 2016-02-08:
> request to support user configurable severity for spoken messages.

Seems interesting, but worried that this really is being requested because the firmware isn't marking severity appropiately. Hence users want to tone town all the talking the firware is causing. Do you have a use case where changing the setting for this is a useful feature? Some example message traffic to show how changing back/forth is helpful?

From my standpoint, adding an option is always a last resort instead a correct default.

sanderux, 2016-02-08:
The request was made in light of people wanting spoken info messages when flying FPV.

DonLakeFlyer, 2016-02-08:
Ah, makes perfect sense, thanks.

closed 2026-10-04 by github-actions[bot]