mavlink/qgroundcontrol #2716: Loop Mission Fail: JumpToItem defaults to ID0 after reload from FC

tubeme, 2016-01-28:
Programming a JumpToItem command in a mission to ID1 repeats: 60 times is accepted by QGC, but after reload from the FC the JumpToItem defaults the ID to "0". So no loop mission is possible.

The Mission file is without extension and I could't attach it. So I archived it in ZIP.
[Loop Mission Fail.zip](https://github.com/mavlink/qgroundcontrol/files/108905/Loop.Mission.Fail.zip)

LorenzMeier, 2016-02-06:
@tubeme Could you try that again with PX4/master? I believe we fixed it there.

tubeme, 2016-02-06:
No it is not working properly. It remembers the ID# -1 and 60 repeats for IDs larger than 2. If ID2 is chosen then it returns ID1 repeats: 0. If ID1 chosen then it returns ID0 and repeats: 0. Do you need any logs or screens?

DonLakeFlyer, 2016-02-09:
This is a QGC bug with the JUMP commands and ids.

DonLakeFlyer, 2016-02-09:
This is going to have to wait for the Planned home position work. It is mired in the PX4/ArduPilot home position in slot 0 difference issues.

DonLakeFlyer, 2016-02-09:
Also just FYI: The IDs are sent correctly to the vehicle and the mission should work fine. The bug is when you read back from the vehicle.

tubeme, 2016-02-09:
@DonLakeFlyer OK then I will fly test it in two days and will report back.

DonLakeFlyer, 2016-02-10:
This is fixed, but there is a problem with previously saved mission files which include a DO_JUMP command. For the last few months (not sure exactly how far back) QGC has been writing mission files with incorrect DO_JUMP indices (off by +1). Given the fact that it has been a longer time that QGC wrote this correctly my fix assumes DO_JUMP indices in a mission file are correct. The brand new json .mission file works correctly.

closed 2016-02-10 by DonLakeFlyer

tubeme, 2016-02-11:
@DonLakeFlyer I just wanted to confirm it works, despite it is closed issue.

http://logs.uaventure.com/view/cv8fabKPAFGMkTNyTi9KQm

DonLakeFlyer, 2016-02-11:
@tubeme Thanks for reporting back.