mavlink/qgroundcontrol #2726: Fixed Wing Mission Planning: Show band instead of path

tumbili, 2016-02-01:
Users are confused with how a plane flies along a path specified by the mission item. They expect the plane to fly by the waypoint as close as specified by the acceptance radius in the mission item.
However, the plane will rather fly along the waypoints in a band which is related to the L1 radius of the controller. It would be useful to indicate this band in the planning view of QGC. This would tell the user that the plane will be flying somewhere inside that band.

LorenzMeier, 2016-02-01:
@DonLakeFlyer @dogmaphobic We could start with a simple spline 200 pixels wide and semi-transparent. We could then go on to refine it based on the actual mission math so it reproduces the path as close as possible.

LorenzMeier, 2016-02-06:
@NaterGator @dagar Would you be interested to look into this? It seems like this is the primary user experience issue left, because users can plan missions that are absolutely not doable by the aircraft and the best-effort based result is confusing them a lot. We need to expose the result on the UI and enforce some basic parameters like waypoint distances.

LorenzMeier, 2016-02-06:
This issue is a result of missing visibility from the user side: https://github.com/PX4/Firmware/issues/3639

DonLakeFlyer, 2016-02-08:
I get the concept of showing the user a better visual of what will really happen in a mission. But in reality I"m concerned this is going to be very difficult and very firmware specific. As a different example what really happens when a fixed wing flies to a loiter mission item with a turning radius? How do you visualize that with a correct path? What if PX4 and ArduPilot do it differently? What about VTOL? Does QGC now need to understand what mode a VTOL is in to show the correct visualization. If you try to get too fancy with correct visuals I'm concerned all hell will break lose.

I think it would be interesting to try to make this work. Since it would be so much of a better user experience. But to really make it work correctly for all mission items types, taking into account firmware differences, I think it will be a significant amount of work. I think this should wait until 3.1

LorenzMeier, 2016-02-08:
It will be a lot of work, yes. I'm wondering if we should enforce the minimum waypoint distance though already for 3.0. That will prevent the primary failure case.

DonLakeFlyer, 2016-02-08:
> I'm wondering if we should enforce the minimum waypoint distance though already for 3.0.

Shouldn't the firmware do this somehow as well? Some sort of warning output when the user tried to fly an invalid mnission. I'm not saying QGC shouldn't do it as well. But doing it only from the QGC side will not be 100% infallible. The only time a turn radius is truly known is when the mission is actually flown. Anything before that is in essence offline editing since the parameters to calc turn radius can be changed or may not be available.

The thing I don't have a handle on right now is in what mode are missions generally created? When people are playing around, I think they are likely done while connected to a a vehicle. But I have a feeling missions which are used to get real work done are done offline which complicates things. Any insight on this?

DonLakeFlyer, 2016-02-08:
Better visuals for vehicle paths is going to have to wait until 3.1. For 3.1 I plan on implmenting custom mission item visuals support. Initially this was only for the item indicators themselves but I can include the path elements as well. It's a fairly involved piece of work which I have already started on for survey support.

This leaves making a call on minimum waypoint distance checks. My feeling is that this should be part of firmware mission validation as an initial starting point. It will be way easier to enforce there since you have full information on the vehicle. It is doable in QGC (and should likely be done in 3.1) but things like offline editing really complicate the situation.

LorenzMeier, 2016-02-09:
@DonLakeFlyer It doesn't really help if you enforce the minimum waypoint distance on the autopilot - its something that has to be done during editing. What you're proposing is an user experience from hell. Missions which fail to upload and you only can figure it out from a cryptic text message. Yuck!

Flight stack neutral offline editing is a fringe use case. I'd even say its non-existent. You can just save the last vehicle editing properties like you safe param meta data and work off that. Then you don't have to deal with any of that. You can even go as far as *requiring* the user to connect his autopilot *once* before you allow offline editing.

Keep in mind that we're talking here about fundamental physical limitations of the airframe. You can't deliver good mission planning experience without factoring that in. So knowing the exact airframe is a given, and we need to work out how to get there.

DonLakeFlyer, 2016-02-09:
> What you're proposing is an user experience from hell.

What I'm proposing is to not deal with this on the QGC side in the 3.0 release, and fix it in 3.1.

> So knowing the exact airframe is a given, and we need to work out how to get there.

I'm already working my way towards that in 3.1 since mapping/survey needs the same thing in order to create a feasible grid flight pattern. My plan is to resurrect the discussion we had on detecting a a new vehicle that QGC has never seen before. Then assign it ID and collect information on vehicle dynamics that can be used for mission. The you could select one of your known vehicle when you start editing a mission.

> Flight stack neutral offline editing is a fringe use case.

Just FYI: QGC has no concept of this any more. In Settings you specify which stack you are targetting when offline editing. Issue is that is not enough, you need full vehicle info (type/dynamics).

DonLakeFlyer, 2016-02-10:
Better visuals for vehicle paths is going to have to wait until 3.1. For 3.1 I plan on implmenting custom mission item visuals support. Initially this was only for the item indicators themselves but I can include the path elements as well. It's a fairly involved piece of work which I have already started on for survey support.

DonLakeFlyer, 2016-02-10:
Moving all of this to 3.1

closed 2026-10-04 by github-actions[bot]