mavlink/qgroundcontrol #2728: Spoken and displayed info from the vehicle

lvale, 2016-02-01:
As the severity of the messages sent by the vehicle is a limited range of options, I believe that on the options of QGC should be a dialog that allowed which severities should be "spoken" and which should pop-up on the screen.

Possibly locking the higher priority messages to behave like it is currently, and only adding the other severities.

LorenzMeier, 2016-02-09:
Its not going to work. The Firmware doesn't have that sort of resolution.

lvale, 2016-02-09:
@LorenzMeier I've recently went through all the messages on ArduPilot to correct the severity they are sent to GCS's, and at the moment all have a well defined set of severities according to what is defined on MAV_SEVERITY at https://pixhawk.ethz.ch/mavlink/

So the suggestion, as discussed on the PX4 meeting is a simple selection box to "speak" the messages according to the severity. Alternatively a second selection box to enable the display of the message on the pop-ups.

Mission Planner when you enable voice commands also has a set of selection boxes to choose what is "spoken" but it's not connected with the severity of messages.
[image]

0	MAV_SEVERITY_EMERGENCY	System is unusable. This is a "panic" condition.

1	MAV_SEVERITY_ALERT	Action should be taken immediately. Indicates error in non-critical systems.

2	MAV_SEVERITY_CRITICAL	Action must be taken immediately. Indicates failure in a primary system.

3	MAV_SEVERITY_ERROR	Indicates an error in secondary/redundant systems.

4	MAV_SEVERITY_WARNING	Indicates about a possible future error if this is not resolved within a given timeframe. Example would be a low battery warning.

5	MAV_SEVERITY_NOTICE	An unusual event has occured, though not an error condition. This should be investigated for the root cause.

6	MAV_SEVERITY_INFO	Normal operational messages. Useful for logging. No action is required for these messages.

7	MAV_SEVERITY_DEBUG	Useful non-operational messages that can assist in debugging. These should not occur during normal operation.

DonLakeFlyer, 2016-02-09:
Why not just have an option to speak all messages? That will be way more understandable to user then trying to have them know what a MAV_SEVERITY means. Tying to select on severity assumes there is really a clear delineation between each separate severity. That's probably not the case as lorenz says, not enough resolution between types.

lvale, 2016-02-09:
@DonLakeFlyer Mav_Severity is what we know that the system uses.

Proposal to be displayed to users would be simple things like

INFO
NOTICE
WARNING
ERROR
ALERT

All those of Critical, Emergency would be always spoken if Voice Alerts were enabled

closed 2017-01-02 by lvale

weiminshen99, 2017-10-10:
Dear Sir,  I am using QGC working wirelessly with a companion computer onboard that autonomously controls the air vehicle in the GUIDED mode. It would be nice to let the human pilot on ground to hear what the autonomous controller is doing in air, especially those important decisions and actions.  
Do we have a way to let QGC to speak those important messages?  Thanks!

DonLakeFlyer, 2017-10-11:
QGC will speak high priority STATUS_TEXT messages which are sent to it. So if you companion computer sent those to QGC, QGC will speak them.

weiminshen99, 2017-10-11:
Thanks. Will give it a try!

weiminshen99, 2017-10-18:
Hi, Don,

Thanks for your kind reply and sorry to bother you again. I have tried many times but cannot figure out how to send out STATUS_TEXT (or STATUSTEXT) messages from my companion computer to QGC. The companion computer is running mavproxy.py and dronekit and I can observe all the STATUS_TEXT messages on my mavlink, but I could not send them out to QGC properly. The following function runs but produce no effects on sending the messages. Could you please give me some hints how to proceed from here? Thanks!

def send_statustext(self, vehicle, severity, txt):
        msg = vehicle.message_factory.statustext_encode(severity, txt)
        vehicle.send_mavlink(msg)

DonLakeFlyer, 2017-10-18:
You're going to need to talk with the Firmware folks about how to set up for a companion computer such that it forwards messages correctly.