mavlink/qgroundcontrol #2743: Do not show the acceptance radius in simple edit mode

LorenzMeier, 2016-02-06:
@DonLakeFlyer Showing the acceptance radius is confusing users. They can also kill mission performance (by setting it too wide or too narrow). This field should never be shown on friendly edit.

[image]

The fact its shown led to user experience issues:
https://github.com/PX4/Firmware/issues/3638

DonLakeFlyer, 2016-02-08:
This did not used to be in friendly edit. Then users asked why it was missing. Then I added it to friendly. Now I'm asked to take it out. Do we really have a correctly informed idea on the users usage of this value? Which users are correct?

LorenzMeier, 2016-02-08:
Users are not actually editing this. Only developers are. You are not showing the radius on the map either. So just remove it.

Please don't get confused by people asking for stuff they had previously in QGC or MP. Both didn't have a user model and the presence of a feature there doesn't mean it made sense.

LorenzMeier, 2016-02-08:
You can also look at it like this: Changing the acceptance radius fundamentally changes the geometry of the mission, because it will lead to a different path taken. So its really confusing if you do it and unless yo have the visuals in the map to indicate this it should not be user facing.

DonLakeFlyer, 2016-02-08:
This is going to be more complicated than just removing it from friendly edit. The issue is that friendy edit also sets the default. If I remove it from friendly edit the default will go away as well which is bad. Also falling back to "Show all values" is too much of a leap for someone that actually wants to set an acceptance radius since "Show all values" just shows Param1-4 and you have no idea which one is accept radius. I'm going to need to restructure some thinks internally and visually to make all of this usable and work correctly.

LorenzMeier, 2016-02-09:
@DonLakeFlyer We need to have to agree on one very basic fundamental principle: Show nothing in friendly edit which is missing the representation on the map. I'm happy to talk you through it on a call, but changing that value changes *EVERY* coordinate in the mission. Its frickin' *DANGEROUS*. So I propose to fix the default dependency on friendly edits, remove it and only re-introduce it with a map representation.

Changing the acceptance radius means you're moving the drone's trajectory. I don't think you nor the user are fully aware of the implications.

NaterGator, 2016-02-09:
Perhaps this is a separate discussion, but isn't the user intuition to expect the craft to "nearly" hit the center of the navigation point even if it achieves an achievement criteria before then? For example in fixed-wing would it not make sense to estimate the time to WP at the achievement met condition and perform a best effort approach to the point?

DonLakeFlyer, 2016-02-09:
> We need to have to agree on one very basic fundamental principle: Show nothing in friendly edit which is missing the representation on the map.

When we get to 3.1 and I have better infrastructure in place for drawing visuals associated with a mission command on the map it will be easier to get closer to this. But whether there is a visual or not will be determined on a case by case basis. I see no reason to hold the 3.0 release because we don't show visuals for acceptance or say loiter radius for example.

Back to acceptance radius:
Mission Planner has acceptance radius as essentially a global setting (visible/changeable by user) which you set once and then it is applied automatically to each waypoint you add. Acceptance radius is not shown in each waypoint editor. Do you think there should be a global setting for the mission visible/changeable by user? Or do you think it should be completely hidden? Given current MP functionality I'm not sure I can get away with completely hidden.

DonLakeFlyer, 2016-02-09:
Plan:
- Remove acceptance radius from friendly edit
- Add Setting for acceptance radius
- Create all waypoints using this global acceptance radius
- Don't show acceptance radius on map

DonLakeFlyer, 2016-02-10:
See #2744, #2773

closed 2016-02-10 by DonLakeFlyer