mavlink/qgroundcontrol #2744: Loiter radius has to be visualised, but only for planes

LorenzMeier, 2016-02-06:
@DonLakeFlyer The friendly edit for loiter waypoints should only show the radius field for planes, not for multicopters. When it does, it should draw a circle at the right scale to indicate where the plane will be loitering. It should also enforce a minimum radius (for PX4:  fixed wing: 1.0f / M_PI * FW_L1_DAMPING * FW_L1_PERIOD * FW_AIRSPD_TRIM), which is the same as the minimum turn distance discussed in https://github.com/mavlink/qgroundcontrol/issues/2726.

[image]

DonLakeFlyer, 2016-02-08:
@LorenzMeier ArduPilot will circle a multi-rotor as well as fixed wing using the specified radius for MAV_CMD_NAV_LOITER_TURNS. For _UNLIM and _TIME it disregards radius for multi-rotor. That seems to make some sense. Do you want to do the same? If not, it causes a problem as to what MAV_CMD_NAV_LOITER_TURNS means for a multi-rotor.

LorenzMeier, 2016-02-08:
It makes no sense at all why one loiter command leads to a circle and one not. Its against the MAVLink spec. We're not going to adopt hackery.

LorenzMeier, 2016-02-08:
That said I'm open to adding a new command which is fully specced and consistent.

DonLakeFlyer, 2016-02-08:
So does this mean that the list of supported commands needs to change based in fixed-wing/multi-rotor?

DonLakeFlyer, 2016-02-09:
With #2755 Loiter Radius is now correctly hidden on multi-rotor loiter command editors. The remainder of the on map visuals will be done in 3.1.

DonLakeFlyer, 2016-02-10:
Final decision:
- Acceptance radius defaults to 0
- Acceptance radius is not show anywhere in UI

DonLakeFlyer, 2016-02-10:
Plas is

closed 2016-02-10 by DonLakeFlyer

szechyjs, 2016-03-22:
Why is there no way to have a multicopter loiter with a radius in qgroundcontrol?

This is possible with ardupilot/mission planner.

http://copter.ardupilot.com/wiki/mission-command-list/#loiter_turns

http://copter.ardupilot.com/wiki/circle-mode/

DonLakeFlyer, 2016-03-23:
@szechyjs Select Loiter Turns from command. Are you not seeing this:
[image]

szechyjs, 2016-03-23:
@DonLakeFlyer I get the following...

[image]

szechyjs, 2016-03-23:
Seems like this is a case of the PX4 firmware is missing the functionality.

https://github.com/PX4/Firmware/issues/3810#issuecomment-200096519

DonLakeFlyer, 2016-03-23:
Oh, I assumed you were using ardupilot firmware. Once PX4 firmware has it I can add.

joe-marrone, 2018-03-21:
I would very much like to have waypoints where a multi-copter loiters and moves in a circular pattern of a specific radius.  I do not see it in QGC 3.3, but is there any plan to do this?  It would be a very useful feature.

Joe

DonLakeFlyer, 2018-03-21:
That would be a firmware feature.

DonLakeFlyer, 2018-03-21:
Wait, do you mean creating individual waypoints in a circular pattern? As opposed to some command which does it automatically?

joe-marrone, 2018-03-22:
What I mean is to go to a waypoint, then do an orbit around that point in
space.  I know this can be done by setting up a bunch of waypoints closely
spaced together, but this is time consuming and laborious, and what if I
want to tell QGC to increase the radius, or go to a different altitude in
real time.  DJI Phantoms can do this in Litchi, and I've had very good
results with it.  So I am looking for the same functionality in PX4 / QGC.
Can it be done, and how?

Joe

DonLakeFlyer, 2018-03-22:
Have you looked at this: 
https://docs.qgroundcontrol.com/en/PlanView/StructureScan.html

and this (Polygon Tools section - Circular polygons):  
https://docs.qgroundcontrol.com/en/releases/stable_v3.3_long.html

Does that works for you?

joe-marrone, 2018-03-23:
Don:  I tried these settings, but the current Pixhawk firmware does not
support them yet.  Too bad, because this is exactly what I want to do.

I posted this issue on the px4 forum, but I have not had a response yet..
http://discuss.px4.io/t/cannot-upload-structure-scan-mission-to-pixhawk/5902

Thanks for your help.

Joe

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

DonLakeFlyer, 2018-03-23:
Support merged into PX4 Firmware master a few days ago: https://github.com/PX4/Firmware/pull/9084. So should be available in developer builds.

joe-marrone, 2018-03-28:
> Support merged into PX4 Firmware master a few days ago: PX4/Firmware#9084

Don,  I just updated to the latest firmware, but it still does not support
pattern/structure.  I am getting the same error messages as before.

Joe

DonLakeFlyer, 2018-03-28:
You installed a developer build, not a beta or a stable? You need developer. I just tested against latest master and it is working. You are using PX4 Pro firmware right?

joe-marrone, 2018-03-28:
Don, I'm not a developer, so please excuse my ignorance.   I am not
familiar with how to build software.

What I did was update firmware from the QGC screen with the Pixhawk
connected to my PC.  Whatever  I got was what was supplied, so I just
assume its the latest version.

Joe

DonLakeFlyer, 2018-03-28:
You don't need to build anything. Let me provide details. There are two main providers of firmware for the Pixhawk series of boards: 
* PX4 Pro
* ArduPilot

Currently only PX4 Pro supports Structure Scan. For each firmware there are multiple versions:
* Stable - Fully tested
* Beta - Next version of Stable which is about to be released
* Developer - The absolute newest changes available. Not fully tested like Stable.

Structure Scan is currently available in the Developer version of the PX4 Pro firmware. If you have only ever used Stable builds, then I might recommend waiting until the next Stable comes out which will then in turn have Structure Scan support in it. If you want to live on the bleeding edge of new functionality here is what you do:

* Go to Firmware page
* Plug your board in
* Select PX4 Flight Stack
* Select Advanced Settings
* Select Developer Build (master)
* Flash the board with that

You now have a version of firmware which supports Structure Scan

joe-marrone, 2018-03-29:
Thanks, Don.  I will consider doing this

Joe