mavlink/qgroundcontrol #2746: Implement drop-down for enums with PX4

LorenzMeier, 2016-02-07:
@DonLakeFlyer MERGED: This adds this param meta for PX4 (note the <values> tag). It would be great if that could be enabled in the UI to drive the drop-downs like for APM. I also have a wish list for a few config UI additions driven by this.

```xml
   <parameter default="157600" name="SYS_COMPANION" type="INT32">
      <short_desc>Companion computer interface</short_desc>
      <long_desc>CHANGING THIS VALUE REQUIRES A RESTART. Configures the baud rate of the companion computer interface. Set to zero to disable, set to these values to enable (NO OTHER VALUES SUPPORTED!) 921600: enables onboard mode at 921600 baud, 8N1. 57600: enables onboard mode at 57600 baud, 8N1. 157600: enables OSD mode at 57600 baud, 8N1.</long_desc>
      <min>0</min>
      <max>921600</max>
      <reboot_required>true</reboot_required>
      <values>
        <value code="57600">Companion Link (57600 baud, 8N1)</value>
        <value code="921600">Companion Link (921600 baud, 8N1)</value>
        <value code="157600">OSD (57600 baud, 8N1)</value>
      </values>
    </parameter>
```

LorenzMeier, 2016-02-07:
@DonLakeFlyer How do I wipe the parameter XML cache? I'm not seeing the new default file I introduced.

NaterGator, 2016-02-07:
@LorenzMeier check the locations listed here for your platform:
http://doc.qt.io/qt-5/qsettings.html#platform-specific-notes

On windows it is located in %APPDATA%\Roaming\QGroundControl.org

closed 2016-02-08 by DonLakeFlyer