mavlink/qgroundcontrol #2749: Add support for MAV_CMD_NAV_VTOL_TAKEOFF and MAV_CMD_NAV_VTOL_LAND

sanderux, 2016-02-07:
Please add support for VTOL takeoff and land commands as specified here: https://github.com/mavlink/mavlink/pull/487/files

PX4 reference: https://github.com/PX4/Firmware/pull/3699

DonLakeFlyer, 2016-02-10:
From looking at the firmware source I think these assumptions are correct:
- [ArduPilot:PX4] Supports coordinate for both takeoff and land
- [PX4] Support heading for takeoff and land
- [ArduPilot] Does NOT support heading for takeoff and land

@sanderux Can you very this is correct?

DonLakeFlyer, 2016-02-10:
Also:
[PX4] - Support VTOL_TRANSITION
[ArduPilot] - Does NOT support VTOL_TRANSITION

closed 2016-02-10 by DonLakeFlyer

sanderux, 2016-02-10:
@DonLakeFlyer as far as i can tell APM indeed does not support VTOL_TRANSITION yet. @tridge ?