mavlink/qgroundcontrol #2765: gazebo cross connections in network

ChristophTobler, 2016-02-10:
We get a lot of cross connections when simulating in gazebo (so e.g. my qgc connects to my friends gazebo simulation). We just tried with latest px4 master and qgc 2.9.2. Is there anything we can do about this? (except unplugging the network connection which is a bit inconvenient)

DonLakeFlyer, 2016-02-10:
@LorenzMeier Can you turn of auto-connect and then change to manual connections with specific IP adreseses in gazebo?

DonLakeFlyer, 2016-02-12:
@dogmaphobic Gus?

dogmaphobic, 2016-02-12:
My guess is that it's using broadcast for all messages. In that case, anyone listening will get it. The idea would be for the *sender* to switch to unicast once it receives a response from a *client*.

dogmaphobic, 2016-02-12:
Different code but the idea applies:
https://github.com/dogmaphobic/mavesp8266/blob/master/src/mavesp8266_gcs.cpp#L102-L107

LorenzMeier, 2016-02-12:
@dogmaphobic It was all the time that way. I can't reproduce this issue with PX4 master with a desktop and tablet. I expect that we're having issues with stale code being used.