mavlink/qgroundcontrol #2776: Blacklist intel drivers for OpenGl

DonLakeFlyer, 2016-02-11:
Need to figure out how to blacklist the intel drivers to use Angle instead of OpenGl.

NaterGator, 2016-02-11:
We can use QT_OPENGL_BUGLIST: http://doc.qt.io/qt-5/windows-requirements.html

DonLakeFlyer, 2016-02-12:
FYI: vasil.petrov.tv@gmail.com has one of these laptops with the intel board which doesn't work

DonLakeFlyer, 2016-02-12:
@NaterGator Do you think you can set this up in the installer? The guy above has an example case you can test against.

NaterGator, 2016-02-12:
@DonLakeFlyer it will be a _little_ bit grosser than that. We likely shouldn't muck with the user's system wide environment variables. 

On Windows I'd advocate we check that QT_OPENGL_BUGLIST is set at launch, and if not respawn via QProcess w/ our custom buglist provided. I'll see if I can get the info necessary from that user to extend the buglist.

DonLakeFlyer, 2016-02-12:
Hmm. Any way to detect the bad drivers ourselves and do the switch internally? I think you can tell the QmlEngine what rendering to use.

NaterGator, 2016-02-13:
Doing anything ourselves would be more of a hack than using the existing Qt code.

The other option is to give users the ability to enforce a particular renderer in the QGC settings page. We can give them Native, Compatibility (ANGLE), or Software rendering. That setting is applied when QApplication is started.

I'll trace through QtBase and find out when the QT_OPENGL_BUGLIST variable is checked. There's a chance we can set it before it is checked by Qt in which case we won't have to respawn ourselves.

tubeme, 2016-02-13:
Year and a half ago There was a version of QGC that stopped working on my Lenovo with nVidia Quadro NVS 140M, despite it is pretty capable VGA. Then I moved to the laptop I'm currently using for GCS. So when the Intel problems started on my current machine I tried again the nVidia Quadro NVS 140M and it works there perfect. So with the progress of Qt strange things happen. Making more complex software brings more problems. Keep it simple and just give 3 icons for the 3 different modes. If the default does not work, users will use the QGC Compatibility and QGC Software Render... you place the icons in the QGC program folder and describe it in the Users manual. Plain and simple. We are probably 1-10% cases that have video problems and this cases are not limited to Intel video. This 3 icons will start QGC with different arguments.

DonLakeFlyer, 2016-02-13:
> The other option is to give users the ability to enforce a particular renderer in the QGC settings page.

QGC crashes out before you get there.

> This 3 icons will start QGC with different arguments.

This is an interesting idea and simpler to implement. Still would be nice to have a runtime check right in main before Qt craps out that will point the user in the right direction.

NaterGator, 2016-02-13:
Another dirty option is to supply a windowless launcher process that acts
as a springboard for launching the main process with the appropriate
renderer.

I'm going to get a first pass done today.

DonLakeFlyer, 2016-02-13:
QGC only crashes out once it starts rendering Qml. So various checks can go in main before any Qml engine is started. Then I think the appropriate renderer can be set for the Qml engine. So you don't need the launcher.

NaterGator, 2016-02-13:
The point is actually moot, here is the stack trace for the OpenGL compatibility check:
[image]

We can easily set the environment variable ourselves with no trickery. Crisis averted, will have a PR momentarily :)

tubeme, 2016-02-13:
BTW on both notebooks and on both occasions I did not have crash but rather hanging.

NaterGator, 2016-02-13:
I'm not surprised. The issue is not Qt getting more complicated but Intel's history of very poor OpenGL implementations on Windows. They have recently improved quite a bit and their drivers are getting less quirky, but in the past they were a mess and badly broken.

NaterGator, 2016-02-13:
@tubeme can you verify if this installer fixes the issue? If not we may need to block GLES too and fall to pure d3d11 rendering, and failing that fall all the way back to software rasterization.

https://ci.appveyor.com/api/buildjobs/mk5f8w8c1xjqw2jc/artifacts/qgroundcontrol-installer.exe

tubeme, 2016-02-13:
@NaterGator No it is not working.... Had to manually set again "angle" to see it working.

NaterGator, 2016-02-14:
@tubeme can you set the environment variable QT_LOGGING_RULES=qt.qpa.gl=true and run again? The log window for QGroundControl should have a bunch of entries for qwindowsopengltester.cpp I'd like to see.

NaterGator, 2016-02-14:
Actually scratch that, I made a silly mistake. Never code groggy! See if this one works for you:
https://ci.appveyor.com/api/buildjobs/g3r1afjkjm4d6ev2/artifacts/qgroundcontrol-installer.exe

tubeme, 2016-02-15:
This latest one does not work either..Here is the latest version output:

[I] at :0 - "Logging ini file directory "C:/Users/Vasko/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Vasko/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/Vasko/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(37.8038, -122.462, 0)"

and then with the argument QT_LOGGING_RULES argument you gave me:

[I] at :0 - "Logging ini file directory "C:/Users/Vasko/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Vasko/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/Vasko/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(37.8038, -122.462, 0)"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"

tubeme, 2016-02-15:
Here is with the ANGLE and QT_LOGGING:

[I] at :0 - "Logging ini file directory "C:/Users/Vasko/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Vasko/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/Vasko/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(37.8038, -122.462, 0)"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"
[I] at :0 - "GPU features: QSet()"
[I] at :0 - "QWindowsOpenGLTester::supportedGlesRenderers GpuDescription(vendorId=0x8086, deviceId=0x2a42, subSysId=0x19f1025, revision=7, driver: "aticfx32.dll", version=8.17.10.1404, "Mobile Intel(R) 4 Series Express Chipset Family (Microsoft Corporation - WDDM 1.1)") renderer:  QFlags(0x2|0x4|0x8|0x20)"
[I] at :0 - "Qt: Using EGL from libEGL.dll"
[I] at :0 - "Qt: Using OpenGL ES 2.0 from libGLESv2.dll"
[I] at :0 - "QWindowsEGLStaticContext::create Created EGL display 0x5a0e740 v 1 . 4"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize -1, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior 0, swapInterval 1, profile  0)"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"
[I] at :0 - "Orphaned log file count 0"

NaterGator, 2016-02-15:
@tubeme thank you very much for the logging output, something really really weird is going on. Your computer is loading ATI Catalyst drivers "aticfx32.dll" on and Intel GMA system. I imagine this is something related to discrete + integrated GPUs. No matter, I think the better course is to unconditionally switch Intel GMA adapters to ANGLE regardless of driver version.

This version should not check the driver version:
https://ci.appveyor.com/api/buildjobs/tlsyattp737mwh9k/artifacts/qgroundcontrol-installer.exe

LorenzMeier, 2016-02-15:
@tubeme Could you give this another try?

tubeme, 2016-02-15:
@NaterGator @LorenzMeier I might be the reason for this mess. Originaly my laptop has 2 GPUs Intel one and ATI Mobile Radeon HD 4670 which broke so i had spare ATI Mobile Radeon HD 5470 and installed it. Probably the 5470 I had is not properly set for Dual GPU work. And it messed all the drivers in the Win10. I fit in this weird 1% of crazy cases that it is hard to create in the lab on purpouse. :) I will try the latest version.

NaterGator, 2016-02-15:
@tubeme weird case or not, I personally have experience with the terrible state of Intel's OpenGL support in the GMA series of graphics adapters. They should be blacklisted.

tubeme, 2016-02-15:
So the first time I ran it it did not work with the following log:

[I] at :0 - "Logging ini file directory "C:/Users/Vasko/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Vasko/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/Vasko/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(37.8038, -122.462, 0)"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "0 -39 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "14.4625 -53.4625 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "27.4625 -66.4625 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:141 - "27.4625"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "27.4625 -111.4625 -84"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 -141.4625 -84"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 3.1953125 216"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 -51.462500000000006 6"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "127.46249999999999 -121.46249999999999 6"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "127.46249999999999 3.1953125 485"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"

For the second time I deleted all instances of the ATI drivers. and then ran it again with SUCCESS:

[I] at :0 - "Logging ini file directory "C:/Users/Vasko/AppData/Local/QtProject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/Vasko/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/Vasko/AppData/Local/cache/QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(37.8038, -122.462, 0)"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "0 -39 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "14.4625 -53.4625 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "27.4625 -66.4625 -39"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:141 - "27.4625"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "27.4625 -111.4625 -84"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 -141.4625 -84"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 3.1953125 216"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "57.462500000000006 -51.462500000000006 6"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "127.46249999999999 -121.46249999999999 6"
[I] at qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayViewWidgets.qml:142 - "127.46249999999999 3.1953125 485"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"
[I] at :0 - "GPU features: QSet()"
[I] at :0 - "QWindowsOpenGLTester::supportedGlesRenderers GpuDescription(vendorId=0x8086, deviceId=0x2a42, subSysId=0x19f1025, revision=7, driver: "aticfx32.dll", version=8.17.10.1404, "Mobile Intel(R) 4 Series Express Chipset Family (Microsoft Corporation - WDDM 1.1)") renderer:  QFlags(0x2|0x4|0x8|0x20)"
[I] at :0 - "Qt: Using EGL from libEGL.dll"
[I] at :0 - "Qt: Using OpenGL ES 2.0 from libGLESv2.dll"
[I] at :0 - "QWindowsEGLStaticContext::create Created EGL display 0x5bd98b8 v 1 . 4"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize -1, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior 0, swapInterval 1, profile  0)"
[I] at :0 - "QWindowsIntegration::createPlatformOpenGLContext QSurfaceFormat(version 2.0, options QFlags(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior 2, swapInterval 1, profile  0)"
[I] at :0 - "Orphaned log file count 0"

NaterGator, 2016-02-15:
Interesting. The both times it selected ANGLE and the second time it still loaded ATI drivers:
> [I] at :0 - "QWindowsOpenGLTester::supportedGlesRenderers GpuDescription(vendorId=0x8086, deviceId=0x2a42, subSysId=0x19f1025, revision=7, driver: "aticfx32.dll", version=8.17.10.1404, "Mobile [...]

Either way we should have all 4th gen Intel GMA adapters defaulting to the ANGLE based renderer.

tubeme, 2016-02-15:
I'm pretty sure it is Win10 issue... I don't see any difference between both settings the standard and angle one in terms of view and performance... So just turn it on for all Intel GMA adapters. Don't even know what will be the difference in the performance. We have no 3D view...

lucasdemarchi, 2016-02-16:
@NaterGator defaulting to angle or sw rendering only on windows, right?

NaterGator, 2016-02-16:
@lucasdemarchi we're operating as described [here](http://doc.qt.io/qt-5/windows-requirements.html#dynamically-loading-graphics-drivers) with the caveat that we automatically use angle for Intel GMA and 4th generation GPUs.

DonLakeFlyer, 2016-02-16:
FYI: I've updated the wiki to be more specific on Windows laptops with Intel integrated graphics drivers.

DonLakeFlyer, 2016-02-17:
Feels good to close this one!

closed 2016-02-17 by DonLakeFlyer

jaxxzer, 2018-05-09:
This is apparently a solution on some machines:
[image]