mavlink/qgroundcontrol #2786: Map tiles can't always be loaded

WickedShell, 2016-02-12:
[image]

The affected area will never look correct again once zoomed out/or in. I arrived at this state by just panning/zooming around the  map at the same time. The console at the time had this:

[I] at :0 - "Logging ini file directory "C:/Users/WickedShell/AppData/Local/QtPr
oject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/WickedShell/AppData/Roaming/QGroundCont
rol.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/WickedShell/AppData/Local/cache/
QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(34.5683, -112.479, 1.669)"
[I] at :0 - "Orphaned log file count 0"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,12,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=12&z=5&s=Gali&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,10,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=10&z=5&s=Ga&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,11,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=11&z=5&s=Gal&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,13,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=13&z=5&s=Galil&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,14,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=14&z=5&s=Galile&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,14,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=14&z=5&s=G&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,12,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=12&z=5&s=Galileo&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,13,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt3.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=13&z=5&s=&scale=2 - server replied: Not Found'"

WickedShell, 2016-02-12:
Did some more pan and zooming (caused some more tiles to not load) this is in the same session (with already bad tiles but at a different spot on the map) Most of the console is the same but the [E] at :0 - "QNetworkReplyImplPrivate::error: Internal problem, this method must
 only be called once." messages might indicate there are multiple ways to create the bug/arrive at the state. 

[I] at :0 - "Logging ini file directory "C:/Users/WickedShell/AppData/Local/QtPr
oject""
[I] at :0 - "Filter rules "qt.network.ssl.warning=false\n""
[I] at :0 - "Settings location "C:/Users/WickedShell/AppData/Roaming/QGroundCont
rol.org/QGroundControl.ini" Is writable?: true"
[I] at :0 - "Mapping cache directory: "C:/Users/WickedShell/AppData/Local/cache/
QGCMapCache55""
[I] at :0 - "home item QGeoCoordinate(34.5683, -112.479, 1.669)"
[I] at :0 - "Orphaned log file count 0"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,12,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=12&z=5&s=Gali&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,10,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=10&z=5&s=Ga&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,11,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=11&z=5&s=Gal&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,13,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=13&z=5&s=Galil&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (8,14,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=
s@177&hl=en-US&x=8&y=14&z=5&s=Galile&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,14,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=14&z=5&s=G&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,12,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=12&z=5&s=Galileo&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (9,13,5) 5 times, givi
ng up. Last error message was: 'Error downloading http://mt3.google.com/vt/lyrs=
s@177&hl=en-US&x=9&y=13&z=5&s=&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (4,6,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=s
@177&hl=en-US&x=4&y=6&z=4&s=Ga&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (4,7,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=s
@177&hl=en-US&x=4&y=7&z=4&s=Gal&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (4,5,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs=s
@177&hl=en-US&x=4&y=5&z=4&s=G&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (4,4,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs=s
@177&hl=en-US&x=4&y=4&z=4&s=&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (5,5,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt3.google.com/vt/lyrs=s
@177&hl=en-US&x=5&y=5&z=4&s=Gali&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (5,3,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt3.google.com/vt/lyrs=s
@177&hl=en-US&x=5&y=3&z=4&s=Ga&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (5,4,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=s
@177&hl=en-US&x=5&y=4&z=4&s=Gal&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (5,7,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt3.google.com/vt/lyrs=s
@177&hl=en-US&x=5&y=7&z=4&s=Galile&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (5,6,4) 5 times, givin
g up. Last error message was: 'Error downloading http://mt1.google.com/vt/lyrs=s
@177&hl=en-US&x=5&y=6&z=4&s=Galil&scale=2 - server replied: Not Found'"
[E] at :0 - "QNetworkReplyImplPrivate::error: Internal problem, this method must
 only be called once."
[E] at :0 - "QNetworkReplyImplPrivate::error: Internal problem, this method must
 only be called once."
[E] at :0 - "QNetworkReplyImplPrivate::error: Internal problem, this method must
 only be called once."
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (36,60,7) 5 times, giv
ing up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs
=s@177&hl=en-US&x=36&y=60&z=7&s=&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (36,59,7) 5 times, giv
[... cut ...]
=s@177&hl=en-US&x=37&y=58&z=7&s=G&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (36,61,7) 5 times, giv
ing up. Last error message was: 'Error downloading http://mt2.google.com/vt/lyrs
=s@177&hl=en-US&x=36&y=61&z=7&s=G&scale=2 - server replied: Not Found'"
[E] at :0 - "QGeoTileRequestManager: Failed to fetch tile (36,58,7) 5 times, giv
ing up. Last error message was: 'Error downloading http://mt0.google.com/vt/lyrs
=s@177&hl=en-US&x=36&y=58&z=7&s=Galile&scale=2 - server replied: Not Found'"
[

WickedShell, 2016-02-12:
Restarting doesn't address the issue, causing me to suspect bad data was cached. I have now lost about half the world of usable map area.

[image]

dogmaphobic, 2016-02-12:
I think google is getting smart about it. I've been noticing the same thing. If you go slowly, you get them. Until we have a *proper* way to get these tiles, stick to Bing for satellite maps.

WickedShell, 2016-02-12:
Does Q support mappbox? And how can I go about clearing my cache? (The directory isn't jumping out at me)

dogmaphobic, 2016-02-20:
This has been fixed by #2846

closed 2016-02-20 by dogmaphobic