mavlink/qgroundcontrol #2794: Support for Bebop 2?

gmaclennan, 2016-02-13:
Does qgroundcontrol support the Parrot Bebop 2 yet? I see support is mentioned in the v2.9.0 milestone, but I don't see any other mention of it in any commits, other than some airframe stubs by @tcanabrava. I am unclear what mechanism I should use to connect to the Bebop if it were to work, I seem to be getting some kind of connection with UDP, but it doesn't read any parameters.

LorenzMeier, 2016-02-13:
Parrot has shipped two units to our developers, which should help.

DonLakeFlyer, 2016-08-15:
@gmaclennan Sorry lost this in the fray of Issues. The BeBop has not parameters so you won't get any back.

gmaclennan, 2016-08-16:
Would be fantastic to see support!

erdnaxe, 2016-08-16:
This can maybe help : https://github.com/Parrot-Developers/bebop_mavlink_ctrl
I packaged the app on Archlinux (AUR) so I'm going to test the app on my Bebop 2 (and I have a xbox controller).

DonLakeFlyer, 2016-08-18:
@erdnaxe How do you start a mission that you sent across?

erdnaxe, 2016-08-19:
By default, the Bebop is in manual control (and no battery status...).
To start a mission you need to send a mavlink file (describing the path) on the FTP server of the Bebop, then tell the Bebop to read the file.
The documentation is here : [http://developer.parrot.com/docs/SDK3/#a-name-common-mavlink-start-a](http://developer.parrot.com/docs/SDK3/#a-name-common-mavlink-start-a)
libARMavlink can be built with only libARSAL so you don't need the whole SDK (I packaged everything on [AUR](https://aur.archlinux.org/packages/?O=0&SeB=m&K=erdnaxe&outdated=&SB=n&SO=a&PP=50&do_Search=Aller)).

erdnaxe, 2016-08-19:
I'm going to see what I can do for video streaming on the Bebop 2. It's a UDP RTP with H264 (you can open that in VLC) but before opening the stream, you need to send a command to activate the stream. I'm going to see if there's a simple way (without the whole Parrot arsdk libs)...

DonLakeFlyer, 2016-08-19:
The 3dr Solo video support works exactly the same way "send command to start stream". https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L383

DonLakeFlyer, 2016-08-19:
The trick will be identifying you are connected to a BeBop so you know when you need to do the special Parrot stuff. The current plugin architecture is based on MAV_AUTOPILOT_* coming back in heartbeat but the BeBop returns MAT_AUTOPILOT_GENERIC if I remember correctly. Let me know if you need any help with what should go where.

MAVProxyUser, 2016-09-07:
Why not use the MDNS advertisement?

gmaclennan, 2016-11-27:
Just checking back in on this, my understanding is from the new milestone label that this is still some ways off? My main interest in this was the ability to do mission planning for the Bebop 2.

DonLakeFlyer, 2016-11-29:
Yes you are correct. That said, I think missions actually work from a standpoint of sending them to the vehicle. It's more about identifying which mission commands are supported. So you are going to be presented with the full set of commands. But the BeBop may not support them and and may respond to some of them in unknown ways.

gmaclennan, 2016-11-29:
Interesting. I will test this out over the next few weeks and report back with findings.

DonLakeFlyer, 2016-11-29:
Just take it slow and be careful :)

yohanboniface, 2017-03-21:
@erdnaxe any update from your tests? :)