mavlink/qgroundcontrol #2800: MAVLink 2.0 testing branch

LorenzMeier, 2016-02-13:
@DonLakeFlyer It would be great to get MAVLink 2.0 supported on a branch:

http://discuss.dronecode.org/t/mavlink-2-0-progress-tracking/23/4

DonLakeFlyer, 2016-02-14:
Is there a built c library repository for it anywhere so I can add it as a submodule like mavlink 1?

jimdsouza, 2016-02-14:
I'm building one at https://github.com/jimdsouza/mavlink/tree/mavlink2-wip_test/generator/C/include_v2.0

DonLakeFlyer, 2016-02-14:
Also where is the firmware MAVLink 2.0 branch?

jimdsouza, 2016-02-14:
Still working on building it https://github.com/jimdsouza/Firmware/tree/mavlink2-test

DonLakeFlyer, 2016-02-15:
Sorry for being dense with submodules. But isn't this way to add the v2.0:
```
Donalds-Mac-Pro:mavlink Don$ git submodule add https://github.com/jimdsouza/mavlink/tree/mavlink2-wip_test/generator/C/include_v2.0 v2.0
Cloning into 'libs/mavlink/include/mavlink/v2.0'...
fatal: repository 'https://github.com/jimdsouza/mavlink/tree/mavlink2-wip_test/generator/C/include_v2.0/' not found
Clone of 'https://github.com/jimdsouza/mavlink/tree/mavlink2-wip_test/generator/C/include_v2.0' into submodule path 'libs/mavlink/include/mavlink/v2.0' failed
```

LorenzMeier, 2016-02-15:
I would copy the ZIP file into place and/or just clone the folder to this location. Not add the submodule again.

jimdsouza, 2016-02-15:
I'm sorry. That's the branch. The repository is https://github.com/jimdsouza/mavlink.git

You can clone this repo and checkout to the "mavlink2-wip_test" branch.

DonLakeFlyer, 2016-02-15:
> I would copy the ZIP file into place

I think I'll just do that. Then I'll let the submodule gods figure out how to do it the right way later.

DonLakeFlyer, 2016-02-23:
Ok. I have QGC compiling with mavlink 2.0. It's working with Mocklink. I built and uploaded firmware from here:  https://github.com/jimdsouza/Firmware/tree/mavlink2-test. But when I talk to the board I'm still getting mavlink 1 messages not mavlink 2 messages. I tested with this:
```
qDebug() << "Message" << message.msgid << (message.magic == MAVLINK_STX_MAVLINK1);
```