mavlink/qgroundcontrol #2803: Implement direct control commands: Takeoff, Land, ...

DonLakeFlyer, 2016-02-14:
Takeoff, Land, Go here, Kill Switch?

DonLakeFlyer, 2016-02-14:
@dogmaphobic Hey icon man: I think this will likely be a DropButton. Any ideas for icons?

dagar, 2016-02-14:
I'm looking forward to "go here". Does PX4 need anything added for this to work?

DonLakeFlyer, 2016-02-14:
Lorenz is specing it all out right now.

LorenzMeier, 2016-02-14:
Here is what I was thinking so far:

Go to location command (MAV_CMD_DO_REPOSITION). This command should lead to a mode switch to POSCTL / GUIDED. Stick activity should override the position command and let the user take over.

param1: velocity: -1: default. > 0: ground speed
param2: Mode: -1: Hold current position, < -1: Loiter radius, 1: go to position (multicopter mode), 2: go to position: Fixed wing mode (for VTOL capable airframes)
param4: Yaw, NaN to indicate no yaw angle change
param5: Lat
param6: Lon
Param7: Alt

DonLakeFlyer, 2016-02-14:
* Using NaN to indicate no yaw change is going to be tricky to handle from a ui side. Will require both a check box and a text field. That's a one-off from anything else we have auto-generated in the ui right now. Would be better if it was something the user could type in as well.
* How do you change altitude only? Maybe speed only as well?
* Same issue as NaN with Mode being a combination value and Enum style field. Hard to represent with a good ui. Would be better to have Mode enum values in one param so you can show combo. Then Loiter by itself shown in a text field.

DonLakeFlyer, 2016-02-14:
Also as far as direct control with PX4 firmware: RETURN_TO_LAUNCH doesn't appear to be implemented in missions just NAV_LAND. Maybe it's implemented only as a direct command? Really need both command in both command/mission mode.

DonLakeFlyer, 2016-02-14:
Also for MAV_CMD_NAV_TAKEOFF you would want options there to go just straight up and also not affect yaw.

DonLakeFlyer, 2016-02-18:
Did this discussion move to somewhere else?

LorenzMeier, 2016-02-18:
No, it just didn't progress yet.

DonLakeFlyer, 2016-02-28:
Include other things like trad heli motor stop from ArduPilot.

DonLakeFlyer, 2016-02-28:
Relate to Issue #2669 Kill switch

LorenzMeier, 2016-04-23:
@DonLakeFlyer Need support for reposition now to finish PX4 implementation:
https://github.com/PX4/Firmware/pull/4347

LorenzMeier, 2016-04-23:
Also moved this back to 3.0 since you already have this in.

LorenzMeier, 2016-05-09:
All implemented.

closed 2016-05-09 by LorenzMeier