mavlink/qgroundcontrol #2807: QGC trashes parameters on connection

kd0aij, 2016-02-14:
Scenario: 
1) Fly without GCS
2) Remove SD card, copy logfiles
3) Power up via USB
4) Connect QGC via wifi
5) discover that SYS_AUTOSTART is now zero, as are all sensor cal params
6) look for means to restore parameters from last flight log;

DonLakeFlyer, 2016-02-14:
What exactly is QGC doing wrong here, and how is it changing parameters?

LorenzMeier, 2016-02-14:
Its sending a param save command as part of the sync process, even during the first sync where no params are changed (since ages). That save could, given the right timing, trash the params if the user shuts down the board right then.

DonLakeFlyer, 2016-02-14:
Not sure why a param save when QGC has changed no params would cause a problem. That would seem to be a firmware problem to me. Anyway, I'll look at making these saves not as stupid.

LorenzMeier, 2016-02-15:
Found the mistake on the PX4 side. Fix going to master.

DonLakeFlyer, 2016-02-15:
Re-opening since i think it still makes sense to at least try to make QGC a little smarter about these uneeded saves.

closed 2016-02-17 by DonLakeFlyer