mavlink/qgroundcontrol #2815: De-select param values in param editor by default

LorenzMeier, 2016-02-15:
The param values in the param editor are selected by default (at least on Android), which leads to the complete field erased on backspace. Its a normal UI convention and its highlighted in blue, but some users reported that this seemed unexpected to them.

DonLakeFlyer, 2016-02-15:
> seemed unexpected to them.

Full select on focus is a pretty standard ui paradigm on almost all operating system uis. The is done on all text fields on all os in QGC. Let me look on tablet specifically though to see if they change that paradigm due to the pop up keyboard thing. If they do, we'll follow. If not, we'll keep the same.

DonLakeFlyer, 2016-02-17:
No this isn't quite right. Focus should only full select if you tab into a field. Not if you click into a field. I'm going to turn auto-full select for now.

DonLakeFlyer, 2016-02-17:
Turned auto-select for mobile

closed 2016-02-17 by DonLakeFlyer