mavlink/qgroundcontrol #2816: Phone / sub-9" operation

LorenzMeier, 2016-02-15:
@DonLakeFlyer @dogmaphobic Our user experience is already not great for 7" tablets. I'm wondering if we should ban them, let people suffer or see how hard fixing the most pressing small-screen issues are.

People trying it will certainly have a bad experience. It seems to take 10" at minimum for the current UI to really work.

dagar, 2016-02-15:
FYI About 55% (66 devices) of the current tablet beta testers are on devices < 10".

Through the play store beta listing I'm getting 3-4 emails every week asking why their phones aren't supported. I wonder if we should add tablet to the name or something.

DonLakeFlyer, 2016-02-17:
Didn't we limit installs to 10" or higher through a store setting?

dagar, 2016-02-17:
It's set in AndroidManifest.xml which also limits the installation of the apk. https://github.com/mavlink/qgroundcontrol/blob/master/android/AndroidManifest.xml#L54

This limits it to roughly 7".

DonLakeFlyer, 2016-02-17:
Ok. I don't have anything 7" to test on. Let me see how cheap a 7" tablet is.

dogmaphobic, 2016-02-17:
I have a Nexus 7. I haven't loaded anything there in ages. I'll take a look when I get a chance.

dogmaphobic, 2016-05-06:
I think this has all been solved by #3333. Closing...

closed 2016-05-06 by dogmaphobic