mavlink/qgroundcontrol #2819: Major work items for 3.1 release

DonLakeFlyer, 2016-02-15:
* Survey support
* Switch to fully/completely/all-the-way/really-I-mean-it-this-time native Qml
* Translation support
* Mavlink 2.0
* Full direct command support: Takeoff, Go Here, Land, Follow Me

NaterGator, 2016-02-15:
Can we take a swipe at optimization here too (#2638)? Optimizing the QML side can/will be a bit tricky but it is probably best to learn and implement best-practices as we go whole-hog on QML.

DonLakeFlyer, 2016-02-16:
Sure, sounds good.

jcooper21, 2016-02-17:
On the land function for APM Plane support is it possible to get the do_start_land command so planes can autoland with APM?

DonLakeFlyer, 2016-02-17:
> s it possible to get the do_start_land command

I added that a few days ago.

jcooper21, 2016-02-17:
Awesome keep up the great work.  The only other main features I see missing that have not been said is vibe monitoring, EKF monitoring, second GPS sat info, Wind speed & direction

DonLakeFlyer, 2016-02-17:
> The only other main features I see missing that have not been said is vibe monitoring, EKF monitoring, second GPS sat info, Wind speed & direction

Can you move those to a new issue and we'll continue discussion there?

jcooper21, 2016-02-17:
Done

jcooper21, 2016-02-20:
I would like to add Orbit to this list

DonLakeFlyer, 2016-02-21:
> I would like to add Orbit to this list

Do you mean like Circle in ArduPilot. QGC already supports it on ArduPilot. Needs to be added to PX4 firmware before QGC can support it then it's a trivial non-major change on the QGC side.

jcooper21, 2016-02-21:
Yes i was reffering to circle. That sounds good.

dagar, 2016-02-28:
How about calibration rotation images or animated models https://github.com/mavlink/qgroundcontrol/issues/1783?

DonLakeFlyer, 2016-11-24:
Closing this out since 3.1 is wrapping up

closed 2016-11-24 by DonLakeFlyer