mavlink/qgroundcontrol #2823: QuadRanger airframe sets wrong autostart

sanderux, 2016-02-16:
When selecting QuadRanger (13008) from standard VTOL, SYS_AUTOSTART is set to 13005

DonLakeFlyer, 2016-02-16:
Can flash again and let me know. I think that was fixed on the firmware side. Built in xml meta data is 13008 but you may still have a flash cached version which is incorrect.

sanderux, 2016-02-16:
This was not through flashing, it was flashed through make and then selecting a different airframe. is that behavior different?

DonLakeFlyer, 2016-02-17:
Nope looks like a QGC bug, not a meta data bug.

DonLakeFlyer, 2016-02-17:
Can you try flashing master using QGC. Quit QGC. Reboot QGC and then try setting the airframe again?

DonLakeFlyer, 2016-02-22:
@sanderux Any update on this?

sanderux, 2016-02-22:
Not yet, will set it up this week

DonLakeFlyer, 2016-03-06:
@sanderux Any update?

DonLakeFlyer, 2016-03-11:
@sanderux ???

sanderux, 2016-03-12:
I just tried to reproduce but failed.
Not sure what happened. I'll let you know if it happens again.

closed 2016-03-12 by sanderux