mavlink/qgroundcontrol #2829: Mission transfer to PX4 trashed via UDP

LorenzMeier, 2016-02-17:
[image]

LorenzMeier, 2016-02-17:
And actually this a usability issue: The real error is shown in a critical message from the system. The critical messages should probably be shown all in the lower message pane:

[image]

DonLakeFlyer, 2016-02-17:
@dogmaphobic Shouldn'y all critical errors be showing up down there? I have noticed sometimes they don't pop. Not sure why.

dogmaphobic, 2016-04-17:
I can't find anywhere in the code base where these *critical* messages are ever sent to that message box. There is one handler within Vehicle.cc that receives it but all it does is to send its own signal, which no one listens to:

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L928

The only messages that are shown in that lower message area are those that are explicitly called. That is, only fully formatted messages are ever sent there. For example, things like these:

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L466-L478

In other words, critical messages coming from vehicles are not currently been sent there at all. I will add code to the signal above (Vehicle.cc#L928) and have the critical messages sent there.

dogmaphobic, 2016-04-17:
Disregard the above. MainWindowInner has a handler for `formatedMessage` (Vehicle.cc#L928 above). Looking...

dogmaphobic, 2016-04-17:
Done in #3241. It was a side effect of some changes Don had done, I assume to prevent the message handler from sending messages before the main window was ready to receive them.

closed 2016-04-17 by dogmaphobic

DonLakeFlyer, 2016-04-18:
> I assume to prevent the message handler from sending messages before the main window was ready to receive them.

I vaguely remember something like that