mavlink/qgroundcontrol #2831: problem connecting PixRacer when USB and wifi are both active

kd0aij, 2016-02-17:
appears to start loading parameters over wifi (slower than USB) and fails to complete
Must disconnect from wifi and restart QGC.
@pkocmoud can you reproduce this?

pkocmoud, 2016-02-17:
When did this start?

DonLakeFlyer, 2016-02-17:
Can get logging out from connection logs: http://www.qgroundcontrol.org/logging. Then we can see what order things are connecting.

kd0aij, 2016-02-17:
[qgc.log.zip](https://github.com/mavlink/qgroundcontrol/files/134855/qgc.log.zip)

DonLakeFlyer, 2016-02-17:
This has full logging which makes it near impossible to pull anything out.. I need the connection logging only. Need --logging:LogManagerLog,MultiVehicleManagerLog. Also the log doesn't look like it starts from boot.

kd0aij, 2016-02-17:
./qgroundcontrol --logging:full 2>&1 qgc.log | tee qgc.log
Is that wrong?

DonLakeFlyer, 2016-02-17:
./qgroundcontrol --logging:LogManagerLog,MultiVehicleManagerLog

Not sure why it doesn't seem to get the beginning of the log.

kd0aij, 2016-02-17:
[qgc.log.zip](https://github.com/mavlink/qgroundcontrol/files/134870/qgc.log.zip)

DonLakeFlyer, 2016-02-17:
The problem is that the WiFi connection comes up before the USB connection does. This in turn causes problems with the parameter protocol with respect to multi-link support. Lorenz, Gus and I just finished discussing the solution for this:
- Have QGC mark that it has seen a bootloader on a USB port
- Then even though it sees the WiFi connection first, it knows that a USB connection is about to come it.
- So it does not connect via WiFi and instead connects via USB only.

What this means: 
- You can only change WiFi module params when "NOT" connected via USB

DonLakeFlyer, 2016-02-18:
@kd0aij Can you try the latest daily build?

kd0aij, 2016-02-18:
parameters load successfully now, thanks. But the artificial horizon doesn't work.
"Comm links" shows only one connection, "default udp link"
Clicking "disconnect" results in a new "default udp link" button and immediate reconnection.
I don't suppose that is the intended behavior?

pkocmoud, 2016-02-18:
> Comm links" shows only one connection, "default udp link"
Clicking "disconnect" results in a new "default udp link" button and immediate reconnection.
I don't suppose that is the intended behavior?

I have noted this for a while using the daily builds. IE it is not recent.

DonLakeFlyer, 2016-02-18:
Don't expect to make any sense out of auto-connected links by going to comm links. There is already an issue in on that. If you want to turn off auto-connect for UDP you need to do that from the General page.

Can you use the mavlink inspector to see what sort of mavlink messages you are getting?

kd0aij, 2016-02-18:
attitude at 94Hz, highres_imu at 50Hz, vfr_hud at 20Hz

kd0aij, 2016-02-18:
numeric roll and pitch display OK

kd0aij, 2016-02-18:
with both "Pixhawk" and "UDP" autoconnect disabled, I can connect properly using either USB or wifi 
So things are much better now.

DonLakeFlyer, 2016-02-19:
Does that fix the attitude indicator? Not sure what you mean by "I can connect properly using either USB or wifi". Why do you need to do that?

kd0aij, 2016-02-19:
artificial horizon still dead.
Regarding "Why do you need to do that?" *somebody* needs to do testing :)

closed 2016-02-19 by DonLakeFlyer