mavlink/qgroundcontrol #2832: New altitude indication feature creep on Desktop ;)

LorenzMeier, 2016-02-17:
@DonLakeFlyer The usability of the altitude indication is terrible. A clear step back. Don't let the feature creep get you!

[image]

kd0aij, 2016-02-19:
I assume there's something obviously wrong with that display. Is it the intended meaning of "home" or the presence of units?

DonLakeFlyer, 2016-02-19:
You can't glance at a tablet sized screen and see the Altitude value easily which is an important value. All fixed now.

kd0aij, 2016-02-19:
does "home" mean "relative to home"? I guess "AGL" was deemed more ambiguous?

DonLakeFlyer, 2016-02-19:
Yes

closed 2016-02-22 by DonLakeFlyer