mavlink/qgroundcontrol #2833: APM Support Freature Request EKF, Vibe, Wing & Amp

jcooper21, 2016-02-17:
A couple of great features to see in Qground control would be vibe monitoring, EKF monitoring, second GPS sat info, Wind speed & direction and current draw.

DonLakeFlyer, 2016-02-17:
I tried getting current from the ardupilot BATTERY2 message but that message didn't seem to be sent. Is there some other way to get current? On GPS sat info, what sort of info do you want (or point me to what in MP already shows that)?

jcooper21, 2016-02-17:
Mission planner shows the sat count number of the second GPS when you click the second mouse button on the OSD and select sat 2 count. That is also how wind speed and direction are displayed. I am not sure how to get the battery current info.

DonLakeFlyer, 2016-02-17:
Ok. I'll add those as properties to the configurable instrument panel section. You know about that right? Does mission planner, show current somewhere. If it can, then QGC can.

jcooper21, 2016-02-17:
Yes. Mission planner shows current right beside where it says voltage on the OSD on the flight data screen. Just to the right of voltage.

jcooper21, 2016-02-17:
Everything I noted is all in the flight data screen on the OSD. No I don't know about instrument panel.  I am having issues trying to change the measurement units in Windows. Is that where I can change measurement units in QGC flight data screen OSD widget?

DonLakeFlyer, 2016-02-17:
Change of measurement units is not yet supported. If you click the instrument panel on the right you can pick additional properties to show:
[image]

jcooper21, 2016-02-17:
That's perfect place for the items I mentioned. Also, how come Windows has units in absolute altitude and Mac is relative, also with speed Mac is in m/s and windows is km/h. I am running APM plane 3.5

DonLakeFlyer, 2016-02-17:
We just recently change from absolute to relative as default. If you had run a build where absolute was default (a period of a few days) it likely saved that as your setting. You'll need to set it to relative manually.

DonLakeFlyer, 2016-02-17:
@LorenzMeier The EKF and WIND ArduPilot dialect messages seems like they should be generic. Should they be moved to common and supported there instead? Or new messages?

LorenzMeier, 2016-02-17:
Yes, they should move to common, and yes it needs new messages. I'll get the discussion going. You will need to support the common message and the APM message in parallel for some time.

DonLakeFlyer, 2016-02-17:
If we get the new messages in, then I can generate the new messages from the ArduPilot dialect messages using my mavlink magic firmware plugin converter thingy. For now I'll plumb the new vehicle object model in.

What about the VIBE thing? Useful to generic as well? Or firmware specific?

LorenzMeier, 2016-02-17:
Yes, generic. Although I honestly believe it might be more efficient factoring it into the estimator status report.

LorenzMeier, 2016-02-17:
https://github.com/mavlink/mavlink/issues/511

https://github.com/mavlink/mavlink/issues/510

DonLakeFlyer, 2016-02-23:
MP screen shots:
[image]
[image]

closed 2026-10-04 by github-actions[bot]