mavlink/qgroundcontrol #2836: APM improvement suggestions

jcooper21, 2016-02-17:
Hi,

After multiple missions on APM Plane 3.5 with QGC I found the following areas that could be improved. 

1. Land waypoint altitude should default to 0m instead of 25m
2. When creating the landing sequence with do start land command the plane glide slope needs to be shown as is comes in for landing. The current setup can result in too steep of autoland. 

3. A confirmation message would be useful when waypoints are uploaded to the craft. 
4.the waypoint lines are orange and the plane line as it flys is orange. Is would be very helpful to have these as different colors to make it easier to see.

LorenzMeier, 2016-02-17:
All these issues apply to PX4 as well, so I renamed this land improvement suggestions.

LorenzMeier, 2016-02-17:
In particular the land altitude not being 0 m is really something we should fix immediately.

DonLakeFlyer, 2016-02-17:
> In particular the land altitude not being 0 m is really something we should fix immediately.

Did I ever tell you how much I like that mission command editors are defined in json? :)

LorenzMeier, 2016-02-17:
Are you now bragging or complaining that I didn't sent a PR yet but are lamenting in issues? 8). I should have a long time ago!

DonLakeFlyer, 2016-02-17:
I was trying to brag and ended up putting my foot into my mouth because altitude default isn't in the default. But it will be soon!

jcooper21, 2016-02-17:
Gents, I forgot to mention the last issue I noticed which is that terrian follow does not appear to work. At least I couldn't figure out how to get it to work.

jcooper21, 2016-02-17:
For normal waypoints shouldn't they have a default altitude of 30m relative or something so you don't rush crashing in the ground from the current defaults.

jcooper21, 2016-02-17:
It would be easiest if the user could set defaults

DonLakeFlyer, 2016-02-17:
> I noticed which is that terrian follow does not appear to work

Terrain following is not yet supported.

DonLakeFlyer, 2016-02-17:
> For normal waypoints shouldn't they have a default altitude of 30m relative

Default altitude for generic mission item which acccepts an altitude is 25 meters.

DonLakeFlyer, 2016-02-18:
@jcooper21 For #2: Do you have a mission file you can attach that I can use to test against with Mission Planner. I assume Mission Planner shows something special here that I need to understand?

jcooper21, 2016-02-18:
Which feature are you looking to test against? What file would you like? Would you like me to screenshot something?

DonLakeFlyer, 2016-02-18:
For #2 talking about a plane glide slope visual

DonLakeFlyer, 2016-02-18:
Screenshot is fine.

jcooper21, 2016-02-18:
You will see the grade in the mission planner screenshot.  It calculates its automatically as you move your waypoints.
[image]

DonLakeFlyer, 2016-02-18:
The altitide display at the bottom shows you altitude differences visually. That doesn't work for this case?

jcooper21, 2016-02-18:
No we need to know the actual slope for a good landing so the slope between two waypoints as the altitude changes. Pulling out a calculator isn't the most ideal.

DonLakeFlyer, 2016-02-18:
Ok, got it. I'll figure something out.

jcooper21, 2016-02-18:
It could be as simple as displaying slopes in the waypoint box so each waypoint auto calcs it and you can see it when you click on the waypoint.

NaterGator, 2016-02-18:
You could store the maximum landing glide slope as a param and then
visually cue if the planned landing approach is compliant or not.

jcooper21, 2016-02-18:
That's a great idea. So when selecting do start land command you enter a glide slope and it's handled that way. Then you drop your waypoints as needed and it says visually if you are beyond your slope parameters.

DonLakeFlyer, 2016-02-18:
> You could store the maximum landing glide slope as a param and then
> visually cue if the planned landing approach is compliant or not.

For survey I'd like to start collecting vehicle characteristics since you need them them to calc things.

jcooper21, 2016-02-18:
I'm interested to see what the survey integration looks like and testing it when your ready. QGC is the best interface yet and I am sure that will continue wi surveying. To me I am hoping you are able to implement something as visually appealing and easy to use as maps made easy have done with Map Pilot for DJI. 

The terrain follow I mentioned earlier I imagine this is a bigger feature to integrate p, so would his be something that would go into 3.1 along with follow me and the rest of those features?

DonLakeFlyer, 2016-02-18:
I think survey is going to require terrain follow in a lot of cases, so terrain follow will likely have to go into 3.1 as well. I've already kicked off some internal discussion on it.

jcooper21, 2016-02-18:
Awesome. I was thinking about glide slope implementation and I came up with this. When you select do start land you would have a box for entering glide slope. Say 25 degrees then that waypoint would turn a different color and make itself back to the distance required for the glide slope from the land location. When you click on the actual do start land waypoint a circle would appear with the land point in the center of it to show the distance. Whatever direction the landing sequence needs to begin at you would just move the waypoint around the circle accordingly. From there if you need to do a technical landing to manuaver around objects then you can just drop waypoints between the landing stay and landing points. The waypoints could have a rule where in the sequence max altitude is based on the glide slope, so it would automatically select the. Max height in the triangle based on distance from the land location. This would be the ideal setup I believe.

DonLakeFlyer, 2016-02-18:
Here is what I have for now, which is gradient in the item status display at the bottom:
[image]

I'll look at your concept and see what I can do. Seems pretty fancy for now, so will likely wait till later.

jcooper21, 2016-02-18:
That looks great and will do the job. I think the suggestion would be something you would want to implement with survey because it required a landing point when the do start land is selected. If it's in with survey I imagine the survey will automatically drop a do start land command and a land point for planes. Seems it would be ideal to be in that package so the landing sequence is built with the survey.

DonLakeFlyer, 2016-02-18:
I think we are now at the point where the basics are done and we can move the fancy stuff for 3.1 release consideration.

dagar, 2016-02-23:
I attempted fixed wing autoland with PX4 today. The process of placing the last waypoint before land, syncing, and then waiting for feedback in the console to know how to reposition (based on FW_LND_ANG) was quite awkward. Could some of this logic be built into QGC?

It would be really nice if the last waypoint was moved based its altitude and FW_LND_ANG possibly with some kind of landing slope display. Any repositioning of that waypoint should be constrained so it can't violate the landing slope requirement. It might make sense when adding a landing waypoint to auto add the last waypoint already at an appropriate distance.

Another minor problem was the inability to adjust the acceptance radius for the last waypoint, which might have helped.

CraigElder, 2016-03-24:
Building on Justin's circle to set the glide slope idea, I suggest adding an indicator for the wind vector and make the circle a dial / knob that can be rotated to place the final way point on the circumference so that the final approach is into the wind.  An option to automatically place the turn to downwind and turn to base waypoints would make the user's life easier as well.
Also imagine a scenario where the aircraft has completed a mission and is on its way back to land but the wind has changed direction during the hour or two the aircraft has been flying. The ability to quickly make adjustments to only the landing sequence at the end of a mission would be very helpful. i.e. changing the heading of the final approach with the knob/ dial and move the down wind and base waypoints to line it up for the landing.

DonLakeFlyer, 2017-03-19:
Discussion in this are should be around the new Fixed Wing Landing Pattern item

closed 2017-03-19 by DonLakeFlyer