mavlink/qgroundcontrol #2840: Handle MAV_CMD_DO_LAND_START correctly

DonLakeFlyer, 2016-02-18:
This is a tricky one becase 0/0 for lat/lon indicate not used. Right now 0/0 ends up moving the waypoint to nowhere land!

DonLakeFlyer, 2016-02-18:
@jcooper21 Is using DO_LAND_START with a 0 for lat/lon values more common than using a specified lat/lon. I've set current default to 0/0.

jcooper21, 2016-02-18:
I am not sure. If it's set to 0 then I am making the assumption the plane skips that point and goes straight to the following waypoint. In that sense yes it is but it's because the landing sequence is not pre build yet in mission planner as I had previously laid how it could be used to simplify the process. Does that answer the question?

jcooper21, 2016-02-18:
How it was specified in today's build of QGC worked. I tested it

jcooper21, 2016-02-18:
To me it would be better if the plane went to the do land start point. It's one less waypoint to drop in the landing sequence.

DonLakeFlyer, 2016-02-18:
I guess I need to understand exactly what this command is supposed to do? Doc says:
"The Latitude/Longitude is optional, and may be set to 0/0 if not needed. If specified then it will be used to help find the closest landing sequence."

What is this concept of "closest landing sequence"?

jcooper21, 2016-02-18:
I am not 100% sure on this but I believe it's if you drop multiple points for very long missions. Having the option for different landing sequences in case you have to emergency Autoland. I would expect this multiple landing sequences is something 0.1% of people use.

DonLakeFlyer, 2016-02-18:
Yes I can sort of see that now from looking at the firmware code. That would seem to say that you would want to set lat/lon most of the time (like you say). What I'm no sure is whether the lat/lon in the message is just a marker to compare against for if you have multiple landing sequences, or whether the plane will actually fly through that point?

jcooper21, 2016-02-18:
How it was setup in today's build works good as far at the Lat/long setting.

jcooper21, 2016-02-18:
The plane doesn't fly through the point right now. It skips the point and goes to the next waypoint in the landing sequence.

DonLakeFlyer, 2016-02-18:
> How it was setup in today's build works good as far at the Lat/long setting.

Yes. The issue in the current build is that it shows Altitude as well, which isn't correct. Also the waypoint lines will go through the point which also isn't correct. I"m trying to fix those two things.

jcooper21, 2016-02-18:
If you look at the free shot I posted in the previous thread the landing sequence waypoint doesn't appear in the flight path. I set the landing sequence point right beside the 12th waypoint.

jcooper21, 2016-02-18:
That was supposed to say screen shot. In the mission planner flight data screen it shows the line going through the do start landing sequence point as well but in the flight planning screen I took the screenshot of it doesn't show the do start land point (it goes invisible when the way point is changed to do start land) They didn't solve the display issue yet either. The plane does not pass directly through the do start land command point.

DonLakeFlyer, 2016-02-18:
Yes, I see that. Also see that there is an Altitude in DO_LAND_START which doesn't really do anything. Maybe I just do the same thing. Don't worry about showing Altitude even though that value has no effect. That will be an easy change.

jcooper21, 2016-02-18:
Ya that will work

DonLakeFlyer, 2016-02-18:
Sorry that last comment was quite correct as far as what I can do with the current system. I'm going to need to special case some stuff to allow for lat/lon to come from map click.

DonLakeFlyer, 2016-02-18:
I think I have the info I need for now. Thanks for the help.

DonLakeFlyer, 2016-03-24:
Talked to Tridge about this. It's more common to set lat/lon with DO_START. So the way it works nows, which is to show DO_START items without the waypoints going through it is about as correct as it's gonna get.

closed 2016-03-24 by DonLakeFlyer