mavlink/qgroundcontrol #2844: Main window handling

LorenzMeier, 2016-02-18:
At the end of MainWindow:: connectCommonActions() I added:

```C++
    // connect QML buttons to menu items so menu item checks match button checks,
    // I'm not sure how to make the QML signal use the new Qt connect syntax because
    // from what I can tell the QML class name includes a generated id that could change
    QObject *toolbar = qgcApp()->findQmlChild("toolBar");
    if (nullptr != toolbar)
    {
        connect(toolbar, SIGNAL(showFlyView()),   _ui.actionFlight, SLOT(trigger()));
        connect(toolbar, SIGNAL(showPlanView()),  _ui.actionPlan,   SLOT(trigger()));
        connect(toolbar, SIGNAL(showSetupView()), _ui.actionSetup,  SLOT(trigger()));
    }
```
 
To make this work I had to add the findQmlChild method and give the MaintoolBar an object name in the .qml file like

```C++
QObject * QGCApplication::findQmlChild(const QString& name)
{
    return _rootQmlObject()->findChild<QObject *>(name);
}
```
 
and 
 
objectName: "toolBar"

NaterGator, 2016-02-18:
Instead of connection to the action triggers, if you chase the rabbit down the hole they eventually invoke methods on QGCApplication that invoke QML methods. 

You could do [http://doc.qt.io/qt-5/qtqml-cppintegration-exposecppattributes.html#exposing-methods-including-qt-slots](this) and bind QGCApplication in QML instead of connecting signals/slots in C++ land. 

I would probably just keep it all in QML and bind to the MainWindow.qml show*View

DonLakeFlyer, 2016-02-18:
In most places like this instead of jumping through hoops and writing a pile of code to track a silly checkmarks. I just removed the checkmarks alltogether since they add little value.

DonLakeFlyer, 2016-02-19:
I think I"m just going to pull the View menu out entirely. It's not worth the effort to maintain. I'll move status bar somewhere else. Anyone object?

NaterGator, 2016-02-19:
It's just a confusing duplication of functionality that is already easily accessible. Banish it.

closed 2016-02-19 by DonLakeFlyer