mavlink/qgroundcontrol #2848: Waypoint display issue

jcooper21, 2016-02-19:
Using a Microsoft surface pro 4 or probably any Windows touch screen I can't scroll down on the screen with my finger out pen. This means once there are enough waypoints to fill up the right side of the screen any additional waypoints cannot be accessed.

DonLakeFlyer, 2016-02-19:
Does the mouse/trackpad work?

jcooper21, 2016-02-19:
I am not sure I would assume so. I don't have the Microsoft keyboard/track pad. Just the pen and my finger.

DonLakeFlyer, 2016-02-19:
Does other scrolling work like parameter list work?

jcooper21, 2016-02-19:
Yes that all works good. I have to press and hold the furthest left menu button. Other than that everything else is good.

jcooper21, 2016-02-21:
This is only in touch screen laptops and 2 in 1s not in tablet mode in windows 10

DonLakeFlyer, 2016-02-22:
I'll find some time to get you some private build to try various changes to figure out what it going on here. Stay tuned.

DonLakeFlyer, 2016-02-23:
@jcooper21 Can you replace your qgroundcontrol.exe with this one: https://www.dropbox.com/s/osbob2k8j6fatg5/qgroundcontrol.exe.zip?dl=0. Keep your old one so you can put it back. 

Then try scrolling the waypoint list with thie new .exe

jcooper21, 2016-02-23:
Don, I am unable to test this week. Will try next week

jcooper21, 2016-02-29:
i just tested it and it seems good

closed 2016-03-01 by DonLakeFlyer

DonLakeFlyer, 2016-03-01:
Great, thanks for helping out