mavlink/qgroundcontrol #2850: Map cache related crash on boot

LorenzMeier, 2016-02-19:
@dogmaphobic Just logging it here - will do a clean rebuild now:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtLocation     	0x00000001081ef4b1 QCache3Q<QGeoTileSpec, QGeoCachedTileDisk, QCache3QTileEvictionPolicy>::rebalance() + 1009
1   org.qgroundcontrol.qgroundcontrol	0x0000000107877a71 QGeoTiledMappingManagerEngineQGC::_setCache(QMap<QString, QVariant> const&) + 2497
2   org.qgroundcontrol.qgroundcontrol	0x0000000107875ef3 QGeoTiledMappingManagerEngineQGC::QGeoTiledMappingManagerEngineQGC(QMap<QString, QVariant> const&, QGeoServiceProvider::Error*, QString*) + 5907
3   org.qgroundcontrol.qgroundcontrol	0x0000000107874582 non-virtual thunk to QGeoServiceProviderFactoryQGC::createMappingManagerEngine(QMap<QString, QVariant> const&, QGeoServiceProvider::Error*, QString*) const + 50
4   org.qt-project.QtLocation     	0x00000001081e76cc QGeoMappingManager* QGeoServiceProviderPrivate::manager<QGeoMappingManager, QGeoMappingManagerEngine>(QGeoServiceProvider::Error*, QString*, QGeoMappingManager**) + 108
5   libdeclarative_location.dylib 	0x0000000112eadbc7 QDeclarativeGeoMap::pluginReady() + 39
6   libdeclarative_location.dylib 	0x0000000112eaec96 QDeclarativeGeoMap::setPlugin(QDeclarativeGeoServiceProvider*) + 326
7   libdeclarative_location.dylib 	0x0000000112eb15ab QDeclarativeGeoMap::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 203
8   libdeclarative_location.dylib 	0x0000000112eb2177 QDeclarativeGeoMap::qt_metacall(QMetaObject::Call, int, void**) + 71
9   org.qt-project.QtQml          	0x000000010947ab6d QQmlVMEMetaObject::metaCall(QMetaObject::Call, int, void**) + 573
10  org.qt-project.QtQml          	0x000000010950f610 QQmlObjectCreator::setPropertyBinding(QQmlPropertyData const*, QV4::CompiledData::Binding const*) + 2880
11  org.qt-project.QtQml          	0x000000010950c8f8 QQmlObjectCreator::setupBindings(QBitArray const&) + 1448
12  org.qt-project.QtQml          	0x0000000109510999 QQmlObjectCreator::populateInstance(int, QObject*, QObject*, QQmlPropertyData const*, QBitArray const&) + 1497
13  org.qt-project.QtQml          	0x000000010950bb57 QQmlObjectCreator::createInstance(int, QObject*, bool) + 4311
14  org.qt-project.QtQml          	0x000000010950a9ac QQmlObjectCreator::create(int, QObject*, QQmlInstantiationInterrupt*) + 668
15  org.qt-project.QtQml          	0x000000010950b4e6 QQmlObjectCreator::createInstance(int, QObject*, bool) + 2662
16  org.qt-project.QtQml          	0x000000010950a9ac QQmlObjectCreator::create(int, QObject*, QQmlInstantiationInterrupt*) + 668
17  org.qt-project.QtQml          	0x000000010950b4e6 QQmlObjectCreator::createInstance(int, QObject*, bool) + 2662
18  org.qt-project.QtQml          	0x000000010950ed32 QQmlObjectCreator::setPropertyBinding(QQmlPropertyData const*, QV4::CompiledData::Binding const*) + 610
19  org.qt-project.QtQml          	0x000000010950c8f8 QQmlObjectCreator::setupBindings(QBitArray const&) + 1448
20  org.qt-project.QtQml          	0x0000000109510999 QQmlObjectCreator::populateInstance(int, QObject*, QObject*, QQmlPropertyData const*, QBitArray const&) + 1497
21  org.qt-project.QtQml          	0x000000010950bb57 QQmlObjectCreator::createInstance(int, QObject*, bool) + 4311
22  org.qt-project.QtQml          	0x000000010950ed32 QQmlObjectCreator::setPropertyBinding(QQmlPropertyData const*, QV4::CompiledData::Binding const*) + 610
23  org.qt-project.QtQml          	0x000000010950c8f8 QQmlObjectCreator::setupBindings(QBitArray const&) + 1448
24  org.qt-project.QtQml          	0x0000000109510999 QQmlObjectCreator::populateInstance(int, QObject*, QObject*, QQmlPropertyData const*, QBitArray const&) + 1497
25  org.qt-project.QtQml          	0x000000010950bb57 QQmlObjectCreator::createInstance(int, QObject*, bool) + 4311
26  org.qt-project.QtQml          	0x000000010950ed32 QQmlObjectCreator::setPropertyBinding(QQmlPropertyData const*, QV4::CompiledData::Binding const*) + 610
27  org.qt-project.QtQml          	0x000000010950c8f8 QQmlObjectCreator::setupBindings(QBitArray const&) + 1448
28  org.qt-project.QtQml          	0x0000000109510999 QQmlObjectCreator::populateInstance(int, QObject*, QObject*, QQmlPropertyData const*, QBitArray const&) + 1497
29  org.qt-project.QtQml          	0x000000010950bb57 QQmlObjectCreator::createInstance(int, QObject*, bool) + 4311
30  org.qt-project.QtQml          	0x000000010950a9ac QQmlObjectCreator::create(int, QObject*, QQmlInstantiationInterrupt*) + 668
31  org.qt-project.QtQml          	0x000000010950b4e6 QQmlObjectCreator::createInstance(int, QObject*, bool) + 2662
32  org.qt-project.QtQml          	0x000000010950ed32 QQmlObjectCreator::setPropertyBinding(QQmlPropertyData const*, QV4::CompiledData::Binding const*) + 610
33  org.qt-project.QtQml          	0x000000010950c8f8 QQmlObjectCreator::setupBindings(QBitArray const&) + 1448
34  org.qt-project.QtQml          	0x0000000109510999 QQmlObjectCreator::populateInstance(int, QObject*, QObject*, QQmlPropertyData const*, QBitArray const&) + 1497
35  org.qt-project.QtQml          	0x000000010950bb57 QQmlObjectCreator::createInstance(int, QObject*, bool) + 4311
36  org.qt-project.QtQml          	0x000000010950a9ac QQmlObjectCreator::create(int, QObject*, QQmlInstantiationInterrupt*) + 668
37  org.qt-project.QtQml          	0x000000010949bed1 QQmlIncubatorPrivate::incubate(QQmlInstantiationInterrupt&) + 225
38  org.qt-project.QtQml          	0x000000010949bca3 QQmlEnginePrivate::incubate(QQmlIncubator&, QQmlContextData*) + 451
39  org.qt-project.QtQml          	0x000000010949962f QQmlComponent::create(QQmlIncubator&, QQmlContext*, QQmlContext*) + 495
40  org.qt-project.QtQuick        	0x0000000108a382f2 QQuickLoaderPrivate::_q_sourceLoaded() + 578
41  org.qt-project.QtQuick        	0x0000000108a37b29 QQuickLoaderPrivate::load() + 233
42  org.qt-project.QtQml          	0x00000001095110e0 QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 544
43  org.qt-project.QtQml          	0x0000000109499189 QQmlComponentPrivate::complete(QQmlEnginePrivate*, QQmlComponentPrivate::ConstructionState*) + 73
44  org.qt-project.QtQml          	0x0000000109497359 QQmlComponentPrivate::completeCreate() + 41
45  org.qt-project.QtQml          	0x0000000109498ae2 QQmlComponent::create(QQmlContext*) + 114
46  org.qt-project.QtQuickWidgets 	0x000000010990115e QQuickWidget::continueExecute() + 238
47  org.qt-project.QtQuickWidgets 	0x000000010990105c QQuickWidgetPrivate::execute() + 620
48  org.qgroundcontrol.qgroundcontrol	0x000000010795c7b9 QGCQuickWidget::setSource(QUrl const&) + 25
49  org.qgroundcontrol.qgroundcontrol	0x00000001079cfe8d MainWindow::MainWindow() + 765
50  org.qgroundcontrol.qgroundcontrol	0x00000001079cfb2c MainWindow::_create() + 28
51  org.qgroundcontrol.qgroundcontrol	0x000000010794424e QGCApplication::_initForNormalAppBoot() + 334
52  org.qgroundcontrol.qgroundcontrol	0x000000010791fcd9 main + 457
53  org.qgroundcontrol.qgroundcontrol	0x000000010784f9e4 start + 52
```

LorenzMeier, 2016-02-19:
@dogmaphobic Clean build still keeps crashing. This issue is for real!

LorenzMeier, 2016-02-19:
Daily build is hosed as well :(

dogmaphobic, 2016-02-19:
Hmm, same place Windows had issues with. All other platforms I tested did fine. I will take care of it in a bit. It's Qt's disk cache size setting. Windows had this crash with values smaller than 100kb.

LorenzMeier, 2016-02-19:
Fixed, thanks!

closed 2016-02-19 by LorenzMeier