mavlink/qgroundcontrol #2860: Validate GPS data quality before showing anything on the indicators

LorenzMeier, 2016-02-20:
I have a user who sent a screenshot with NaN on the GPS position and altitude. Not sure if the onboard filter blew up and is to blame or if its a QGC issue. When users have really bad GPS reception indoors all horrible things happen.

We probably need a concept of "good GPS lock" in QGC and also allow flight modes to be disabled in the dropdown when not available. The indicated altitude will also not be correct, etc.

DonLakeFlyer, 2016-02-21:
QGC already has a concept of "good GPS lock". The Vehicle location is not valid until we get 3d fix. But right now the Altitude property can be displayed at any time. So that needs to be changed. I assume Altitude was coming from Instrument Panel? Where was a NaN being shown from GPS position. We don't show that numerically anywhere that I remember.

DonLakeFlyer, 2016-02-22:
Show --.-- is no gps lock for values which from from gps.

DonLakeFlyer, 2016-02-22:
I can't implement this because currenty groundspeed and altitude come from both the GPS and VFR_HUD messages. Since VFR_HUD message sends 0's for these when no GPS lock there is no way for me to tell when I should change these to ---.--- visual. Not sure if I should disconnect groundspeed/altitude from VFR_HUD message.

DonLakeFlyer, 2016-02-23:
@LorenzMeier Any comment on what to do here?

LorenzMeier, 2016-02-23:
Not yet. Trying to come up with a structural fix.

DonLakeFlyer, 2016-05-13:
@LorenzMeier ?

LorenzMeier, 2016-05-13:
Well, that's enough complex to warrant a call.

DonLakeFlyer, 2016-05-13:
Let me do a full writeup on where all the Vehicle object model values are coming from. Then we can talk.

closed 2026-10-04 by github-actions[bot]