mavlink/qgroundcontrol #2861: Android deployment

LorenzMeier, 2016-02-20:
@dagar Would be great if we could finish the automation:

Todo:

- [x] assemble apk on travis-ci
- [x] tracis-ci encrypted android keystore for release apk
- [ ] deploy continuous builds to google play beta
- [ ] deploy release tags to google play production
- [ ] run unit tests in emulator?

LorenzMeier, 2016-02-20:
Last APK upload was on Feb 15 - what failed?

LorenzMeier, 2016-02-20:
It turns out the upload still seems to work, but the binary does not become visible in the dev console nor does it get pushed to users.

dagar, 2016-02-20:
The problem is the google play upload script, or maybe just the credentials it's using. I'm not sure yet.

      File "/home/travis/build/mavlink/qgroundcontrol/tools/google_play_upload.py", line 96, in <module>
        main()
      File "/home/travis/build/mavlink/qgroundcontrol/tools/google_play_upload.py", line 50, in main
        credentials = client.SignedJwtAssertionCredentials(
    AttributeError: 'module' object has no attribute 'SignedJwtAssertionCredentials'

dagar, 2016-02-21:
The oauth2client authentication method I used in the upload script was removed in version 2. Forcing pip to install an older version for now. https://github.com/mavlink/qgroundcontrol/commit/db1987ac624fd1c28704301ca969f160d93fdd65

https://github.com/google/oauth2client/issues/401

dagar, 2016-02-21:
On second thought forcing an old version seems like a bad long term solution, so I updated the script to use the new api. Auto upload to Google play has been restored.

@DonLakeFlyer @LorenzMeier when you're ready to release a production version of the Android app (v3 I'm guessing?) give me a couple hours notice and I'll modify everything so tagged builds end up in production.

dagar, 2016-02-21:
Running tests in the emulator on travis-ci is doable, although a little slow. Main blocker at this point is determining how to actually run them through adb.

DonLakeFlyer, 2016-02-21:
Daniel can we release both Stable and Daily with Daily continuing on as Beta?

Seems like I remember Gus saying he android simulator was crap. So not sure running unit tests in it would help much. Maybe iOS one if better and we could get mobile unit test coverage from that. I can give the android one a try to see what happens.

dogmaphobic, 2016-02-21:
Android simulator is what it says. An ARM emulator running native code (the same code that is deployed). It's slow and unreliable. At least on Mac OS, maybe it runs better elsewhere.

iOS simulator is a different build than a regular deployment build. It builds the app using native code/compilers and runs in a simulated iPhone/iPad environment (on "Intel" code).

dagar, 2016-02-21:
@DonLakeFlyer builds off of master will continue going to the beta channel. Anything tagged will go to production. There will be only one play store entry, but if you opt into the beta channel you'll get the continuous build.

While we're on the subject we should think about updating the description and possibly the name to make it absolutely clear it's tablet only for now. When accessed from the web interface the play store will just say incompatible with your device if the screen it too small, but not tell you why.

Both android_armv7 and android_x86 are possible for testing.

DonLakeFlyer, 2016-02-21:
Is there a way to have Stable and Beta channels have different device requirements? We've been talking about Stable be 9" and above and then open up Beta to anything.

dagar, 2016-02-21:
That's set by AndroidManifest.xml
https://github.com/mavlink/qgroundcontrol/blob/master/android/AndroidManifest.xml#L53

So for the first production build you could leave it as is (tablets only),
then change android:normalScreens in master to enable phone support. If you
wanted to maintain this situation longer term we could look into setting it
during the build.

commit 2c14826c (2016-02-28): travis-ci push tagged builds to google play prod

-closes #2861
closed 2016-02-28 by dagar