mavlink/qgroundcontrol #2873: HDOP signal strength indicator needs updating

DonLakeFlyer, 2016-02-23:
In discussions with a user they were saying the signal strength indicator for GPS wasn't working properly. This was because the current boundary values on HDOP for percentages don't match what users want:
```
      if (hdop < 3)
            return 100
        if (hdop < 6)
            return 75
        if (hdop < 11)
            return 50
        if (hdop < 21)
            return 25
```

In discussions with @gervaisK we came up with this instead:
100% - good to fly - HDOP <= 1.0
75% - ok to fly - HDOP between 1.0 and 1.4
50% - careful, orange - HDOP between 1.4 and 1.8
25% - don't fly, red - HDOP > 1.8

Would like some more discussion to tweak these...

DonLakeFlyer, 2016-02-24:
Changing this to GPS signal strength. I think the real issue is whether it is possible to provide the user with some sort of "signal bar" base indicator for GPS quality. In discussion with various folks the concept of using only HDOP for this bar to indicate "flyability" may be too simplistic.

From a "flyability" standpoint it may require taking into account other factors:
- HDOP
- Sat Count 
- EKF info?

Other info I've gathered:
- 7 sats is green, 6 yellow, 5 bad
- Or Falcon UAV preflight script: sats > 6, hdop < 4

The real question is what are these bars supposed to represent? I know they currently represent HDOP but I don't think there is a concensus as whether you can translate that into a "bars" style graph. And evern if you do, what does it mean?

WickedShell, 2016-02-25:
<= 6 is usually quite poor with u-blox has been my experience and I don't recommend it. I've also never struggled to have 8+ outdoors however so I don't think that is a big concern in general.

Satellite count like using hdop is a poor representation of the quality of you're GPS as it doesn't account for multi path or poor reception. The two are also directly correlated (DOP is of course related to the relative positions of the satellites) adding satellites will improve the DOPs,

The ekf is more of a consistency check then anything else, but also has the potential to be the most helpful.

A useful metric is the jamming indicators/confidence level that u-blox can report, and this does correlate to sources of interference/degraded performance, however the threshold is different for every install, which makes it impossible for a GCS to provide a universal/reasonable default so we don't even report this over the air at the moment from APM (although we do log it).

Satellite CNo is a good metric, but there isn't a convenient way to send it over the air, and it consumes a fair lot of bandwidth to send all of the unique values for it. I'm not sure if averaging it would be a good solution or not...

An obvious one (again not really helping with the bar value) is to throw warnings for fix types (IE no fix/1D/2D would all be red 3D and up would be green. (On APM if you set the EKF to ignore the height value from the gps then 2D would be acceptable to be green but only in that scenario, and even then it's probably not going to be good (as the GPS in general should be finding a 3D fix))

From the APM side (again specifically u-blox related) I can change the health bit of the GPS if I haven't completely configured/verified GPS settings, or if I'm seeing an interruption in the GPS's ability to maintain it's update rate, but this doesn't really help with the bar.

GervaisK, 2016-02-25:
A green indication for a 3D fix doesnt help either, like this example shows:

[image]

"7 sats is green, 6 yellow, 5 bad"

While that might be appropriate for planes (I don´t know) and pretty common before 2014 with u-blox 5,6 , this doesn´t apply to Copter (anymore). With a (nowadays very cheap and popular) u-blox 8 series assy (Dual GNSS with concurrent GPS / GLONASS activated) assy with a passive 25x25x4m patch,  you´ll get this low Sat Count, even indoors if one window is in the room (multipath as well) or within a forest, nevertheless unusuable with Flight Modes like PosCtl. 

So a green bar, Sat count related, or 3D Fix oriented, would result in misjudgement of the real situation. 

Within an ideal environment like on the sea, with low elevation mask and Sat positions placed around me, same Sat Count could result in lower HDOP and good enough results as well. With my big Dual GNSS (GPS and GLONASS since 2014 35x35x6mm patch, NEO M8N mounted onto a Mast) , I experience perfect Loiter (PosCtl.) >=12 Sats with HDOP <=1. Both of them are indicating Green for me.  

Within partly shaded environment, I won´t start with a HDOP much worse than that.(don´t care too much for SAT Count) Most likely to experience glitches. Btw. just for reference : AC sets the default HDOP for Pre-Flight at 1.4 (but its adjustable)

Although this experience is mainly based on APM copter usage from 2.91 till now, I didn´t have different results with latest PX4.

LorenzMeier, 2016-02-25:
@DonLakeFlyer To wrap up: Do the change to the mapping. Move the sat count into the details because it confuses users. If indicating any number, use HDOP.

PX4 now does the same as DJI and APM in terms of reporting:
https://github.com/PX4/Firmware/pull/3839

LorenzMeier, 2016-02-25:
Best indication of signal quality would be to show a circle like on Google maps on the map. Then people understand that a large circle means its not safe. I'd focus on that.

GervaisK, 2016-02-25:
I agree , very valuable at higher zoom levels , where a m/ft dec. out would not help that fast. I enjoy this at Garmin Marine Plotters .

WickedShell, 2016-02-25:
Will the circle reflect the actually meter level of DOP (IE the accuracy of
the GPS * DOP) if so can the user adjust their device accuracy or will a
number be hard coded for that?

GervaisK, 2016-02-25:
My understanding of the circle is, that it reflects eph value .  

Btw. how do you want to "adjust" device accuracy ?

DonLakeFlyer, 2016-02-26:
> Do the change to the mapping.

I don't have any concensus on those mappings. Most ardupilot folks said they fly with 3.0 or less. From that discussion those mappings are too aggressive.

LorenzMeier, 2016-02-26:
Less is more accurate- so the mapping would fit. How about starting with that?

closed 2016-02-26 by DonLakeFlyer