mavlink/qgroundcontrol #2875: RTK GPS base station support

LorenzMeier, 2016-02-24:
@DonLakeFlyer @dogmaphobic We have a joint project with u-Blox on integrating their RTK solution. I'd be interested to turn a QGC tablet into a base station and transmit the correction data via MAVLink.

I need to work out the interface details.

https://github.com/mavlink/mavlink/issues/512

LorenzMeier, 2016-02-24:
Got some info: Works with a local M8P with this config:

UBX-CFG-MSG activate message F5-05 RTCM3.1 1005 
UBX-CFG-MSG activate message F5-4D RTCM3.1 1077 
UBX-CFG-MSG activate message  F5-57 RTCM3.1 1087 
UBX-CFG-PORT set RTCM 3 as Output Protocol        
UBX-CFG-TMODE3 activate Self-Survey-In or Fixed-Position-Mode

LorenzMeier, 2016-02-24:
I will break out the PX4 GPS lib as a standalone library so we can reuse it on both ends to talk to the modules: https://github.com/PX4/Firmware/tree/master/src/drivers/gps

LorenzMeier, 2016-02-24:
@dogmaphobic @DonLakeFlyer If you can choose between Wifi and Bluetooth, which one would you prefer?

DonLakeFlyer, 2016-02-24:
> I will break out the PX4 GPS lib as a standalone library so we can reuse it on both ends to talk to the modules

Will this work for the big honking RTK things as well?

dogmaphobic, 2016-02-24:
> If you can choose between Wifi and Bluetooth, which one would you prefer?

They are not exactly comparable. Assuming this is just for near distance configuration, I would still prefer WiFi even if it means losing Internet connectivity while at it. BT is way too finicky and unreliable.

LorenzMeier, 2016-02-24:
@DonLakeFlyer The u-Blox solution will commoditize RTK, at least for the drone space. I don't think that the existing solutions will remain as high priced, and only the ones adjusting quickly will survive. U-blox has a great reputation risk putting out something half-baked, so I expect it to be good and affordable (could still mean couple hundred, but not couple thousand).

rcorbari, 2016-03-03:
And this is a good news @LorenzMeier :)

kvenux, 2016-03-24:
@LorenzMeier I have the same idea too.
Recently I intended to use two ublox m8n modules(one mounted on pixhawk as rover, the other one connect to GCS as base station) to realize acceptable RTK performance.
I want to use rtklib, actually rtkrcv in app folder to get it right to work(two gps raw data path from serial\TCP). But I cannot get ubx raw data from mavlink. One way to get raw data is GPS passthrough, which can only used under usb connection. 
How can I carry on? Any ideas?

bchristal, 2016-07-29:
Hi,
Just a comment for RTK integration. When using RTK for mapping you do not need precision UAV navigation, rather precise camera locations. If the RTK system is updated at 5 - 10 hz, camera events will fall between GPS update events. This will not provide the accuracy needed for mapping. The RTK system should be capable of polling the exact moment the camera event happens.

Thanks!
Brian

kikislater, 2016-08-18:
Hi @bchristal , 
In proprietary system like ebee or other (ebee is just 1hz) they are using cam sync and interpolation between GPS update events. It began camera update a long path.
Emlid reach purpose is to do the same https://community.emlid.com/t/3dr-solo-with-reach/1846/25

DonLakeFlyer, 2016-08-18:
Need to add android support

bchristal, 2016-08-18:
@kiki. Here's the answer I got from Reach on events: https://community.emlid.com/t/reach-refresh-rate-for-camera-events/2866
Also, looks like it will be an issue with the uBlox as well: https://forum.u-blox.com/index.php?qa=5242&qa_1=c94-m8p-time-mark-documentation&show=5247#c5247

Hopefully we can get precise events with interpolation added as a feature.

kikislater, 2016-08-19:
I think reach is doing interpolation also. It's not really clear. 
If you have values, interpolation is not so difficult to do in post like in GIS.

So I think a workflow for uav mapping rtk should be : 
1) data acqusition (pictures + gps raw)
2) GNSS data correction (getting gnss solutions with rtklib and local permanent base station)
3) Camera sync with gnss solutions

DonLakeFlyer, 2017-04-13:
RTK is working now with ublox

closed 2017-04-13 by DonLakeFlyer