mavlink/qgroundcontrol #2882: Set default font for asian language systems

LorenzMeier, 2016-02-25:
@DonLakeFlyer @dogmaphobic Note that most of the fonts use a seriph font, but the compass fonts look good. Its doing something different, and we should force that for all text, at least on non-western language settings.

[image]

DonLakeFlyer, 2016-02-25:
Windows fonts, sigh..

dogmaphobic, 2016-02-25:
Be ready for a nightmare. A typical case of *Weltschmerz*. Things that took 8 characters and fit nicely on a button? Be ready for a 40-character counterpart. Non Roman alphabets, right to left writing, completely different sentence component position, it's crazy making.

DonLakeFlyer, 2016-02-25:
You know, I wasn't thinking about .json files. There are strings to translate in there. That won't fit into the normal Qt translation system. Hmm...

bys1123, 2018-11-11:
[image]
@dogmaphobic @DonLakeFlyer In chinese maybe is not that complicated, could it add default font Chage this? From windows 7, chinese language environment is default using a sans-serif font called "微软雅黑"（Microsoft Black）.

dogmaphobic, 2018-11-11:
That's very strange. QGC has its own, Sans Serif font. The screen shot above shows something else. Here is where QGC loads its own, default font:

https://github.com/mavlink/qgroundcontrol/blob/master/src/ui/MainWindow.cc#L117-L123

And this is what it should look like:

[image]

bys1123, 2018-11-12:
@dagoodma Seems like only in Windows it's not right. (But, lots of developer is still using Windows in china. )

closed 2025-07-04 by HTRamsey