mavlink/qgroundcontrol #2884: Offline maps usability

DonLakeFlyer, 2016-02-25:
Quick notes I took on using Offline Maps:
* Use the same logic as Plan view for initial position of map
* Better real estate use? Take up the whole space with the map and float controls over it? Buttons to right side, previews could go there as well?
* Default map type to best variation of General Setting
* Naming of "Default tile set" was confusing to me at first. Then I clicked it and saw the description was "System wide tile cache". Then I understood. Maybe the strings for name and description?
* What happens if tel sets they overlap? Do they share?
* Show me where the tile set is when I click on it after I've already made it.
* I don't think most people will understand zoom level so I think Min/Max preview should be on by default

dogmaphobic, 2016-02-25:
Yep, this is what I needed. I put there the basics to get it up and running. Once people started using it, I would know what needed to be done.

First of all, I've added a new MapManager class that should be used across all modules that use maps. It gives a better list of available maps and whatnot. My intent is to consolidate that functionality with the one provided today by `FlightMapSettings`.

> What happens if tel sets they overlap? Do they share?

Yes, the tile database is separate from the *Tile Set* database. Once a tile is cached, it's unique. It only gets deleted when the last set using it is deleted. Which means, the *Default Set*, or everything.

> Show me where the tile set is when I click on it after I've already made it.

The code is already there. It's not working because it's not supported within a QtQuickWidget.

DonLakeFlyer, 2016-02-25:
> Once a tile is cached, it's unique.

Awesome!

closed 2024-09-22 by HTRamsey