mavlink/qgroundcontrol #2890: Value assigned @ Widget don´t appear @ QGC Android , Value Widget not scrollable

GervaisK, 2016-02-25:
[image]
[image]

DonLakeFlyer, 2016-02-25:
Are you running latest Daily? I made it scrollable a few days ago I believe. I also made more space available to the values in the middle. But it's still the case on really small screens there may not be enough room to fit values in the middle there. I haven't figured out what to do about that yet. Right now QGC is not optimized well for below 10" tablets.

GervaisK, 2016-02-25:
Latest apk from here 
https://github.com/mavlink/qgroundcontrol/releases/tag/v2.9.4

i don´t mind if its not optimized, Parameters are scrollable, which is fine

DonLakeFlyer, 2016-02-25:
That's a Stable. It won't be fixed there. Try using Daily Builds instead. Then you'll see the changes.

GervaisK, 2016-02-25:
Latest is a stable ? Sorry new for me, where do I find daily ?

DonLakeFlyer, 2016-02-25:
http://www.qgroundcontrol.org/downloads You can sign up for auto-install of Daily Builds by joinging the beta group. The release tags in Git Hub are only Stable builds. Daily Builds are newer than Stable builds.

LorenzMeier, 2016-02-25:
@GervaisK The best is to sign up for the beta: http://www.qgroundcontrol.org/downloads#daily_build

@DonLakeFlyer Its not daily builds. Its master with 30-90 min delay only.

GervaisK, 2016-02-25:
I am assigned .

GervaisK, 2016-02-25:
No not today. It did not come. Thats why i asked for the apk.

DonLakeFlyer, 2016-02-25:
> Its not daily builds.

I know. But most users have no idea of the concept of "master". Hence the wording on the sites uses "Daily" term. Close enough.

GervaisK, 2016-02-25:
The daily works with win installation.

GervaisK, 2016-02-25:
But Android is leading to the playstore (after assigned as beta tester) and announces to load later on. But that did not work. So,thats why I searched for and figured Latest is Maser (like @ APM).

GervaisK, 2016-02-25:
> On Android there is no need to install anything yourself. Just use the beta and keep it flowing.

Btw. the version I installed, w/o scrolling is from 25th

DonLakeFlyer, 2016-02-25:
> announces to load later on

Can you do a screenshot? You should just be able to install latest master from there.

GervaisK, 2016-02-25:
yes moment.
[image]

Its telling me to wait, thats what Lorenz meant with the delay.

DonLakeFlyer, 2016-02-25:
I'm loading it up on my 7" tablet right now for testing.

DonLakeFlyer, 2016-02-25:
Tried very latest master build on my 7". It scrolls for me. If your instrument panel doesn't have a little gear icon to the right of the hud instrument you aren't running the latest build.

It's still not quite right though since it scrolls under the dialog title. But that's just cosmetic.

DonLakeFlyer, 2016-02-28:
@GervaisK Any update on this?

GervaisK, 2016-02-28:
Yes, Don. Its a matter of my 5.5" screen, cutting the right side of QGC.

DonLakeFlyer, 2016-02-28:
Ok, so we're left with real estate issue on the instrument panel and ..

> Still I would like to be able to turn off the pop up screen and page change, when QGC notices, that there´s a problem w. parameter update.

Do you have screen shot for that? Are you talking about missing parameters?

GervaisK, 2016-02-28:
I will provide a screenshot, next time it happens. QGC is claiming that it´s not able to load paramaters but thats not important during flight as long as battery sats HUD and map view is still present. But it switches to anonother page, and thats what I don´t like. Can live with pop up warnings e.g. manual control lost, but not with page change.

DonLakeFlyer, 2016-02-28:
If QGC says it is missing parameters you should not disregard it. This means something is quite wrong. Either a firmware bug, or possibly your telemetry connection is crap, which is also a really bad thing. That is just a pop up, not a page change though.

Are you talking about when QGC connects to a Vehicle which has an incomplete setup and it takes you to Vehicle summary to finish configuring?

GervaisK, 2016-02-28:
I´m not talking about a setup prob.Would care less. This page change happens during a perfect flight.What I do have, are quite a few diconnects, but it mostly regains connection. If it doesn´t,I unplug and and plug in again. I´m using a genuine 3DR v2 433MHz telemetry kit. btw. Lost manual control appears sts., although there´s nothing lost. But thats a pop up (I don´t mind anymore) you can see e.g. in this video @ 0:29. The prob is, I can´t do screenshots or pics during flight., have to land immedeately and take a pic if it´s still there.

DonLakeFlyer, 2016-02-28:
Lost communication doesn't quite mean you are disconnected. It just means QGC has stopped getting heartbeats from the Vehicle. So that would indicate a telemetry problem.

What exactly does that pop up say from the video?

DonLakeFlyer, 2016-02-28:
I'd bring up the lost manual control thing on the firmware side.

GervaisK, 2016-02-28:
> What exactly does that pop up say from the video?

Sorry, There were several pop ups , which I can´t remember because I´m concentrated on flying and on checking different behavior due to parameter changes, and since I don´t have any probs, I cared less for the them. The unexpected page change sts. is an issue though. Next flights I´ll do a write up.

DonLakeFlyer, 2016-03-01:
Check out Issue #2916. Making progress.

DonLakeFlyer, 2016-03-06:
@GervaisK Did you ever figure out what the page switch was?

GervaisK, 2016-03-06:
I did not forget about this, but got stuck indoors the last 5 days. Back probs.

GervaisK, 2016-03-09:
Value widget assignment is scrollable @ 5.5" Landvo L800 and assignable as well now ! Now an update button appears@google store which makes it easier to ensure latest beta. Thx !  Page switch did not appear again so far, but still grounded due to FM selection issue.

closed 2016-03-09 by GervaisK